Ultimaker/CuraEngine #444: Thin Wall Detection: 2.4B2

CCS86, 2017-01-26:
Hi guys, 

I'm seeing some variability in thin wall detection. I have a part with two identical thin wall sections. One is printed as a single bead at 0.4mm line width:

[image]

[image]

But the other side will not print, even with the line width reduced to 0.2mm. Is the machine nozzle size used during calculations here at all? Is it just an artifact of build plate position and rounding?

[image]

CCS86, 2017-01-26:
[SafetyTop.stl.txt](https://github.com/Ultimaker/CuraEngine/files/732514/SafetyTop.stl.txt)

CCS86, 2017-01-26:
Okay, I answered one of my questions. Leaving the part fixed, and editing the machine nozzle size to 0.3mm, and suddenly it prints. I can actually bump the line width up to 0.34mm and still get both sides printing.

This begs two questions: 

Why does one side print at 0.4mm machine nozzle and line width, while the other doesn't?

Why is machine nozzle size used in calculations, not line width? It seems to me that the machine setting should really just be used to compare and check line width settings, while line widths drive all path generation.

Thoughts?

CCS86, 2017-01-26:
Added actual wall width to original post.

Ghostkeeper, 2017-01-26:
> Why is machine nozzle size used in calculations, not line width?

I've just checked and it seems we use the nozzle size still for the merge infill lines algorithm. I wouldn't think that has an effect on the walls though... The only other place we use nozzle size is for the g-code header of Griffin, but that has no effect on the layer view you see there.

BagelOrb, 2017-01-27:
[image]
[image]

The model is not the same on both sides.

closed 2017-01-27 by BagelOrb

CCS86, 2017-01-27:
Doh! Sorry for being mistaken about the model's symmetry.

It still doesn't explain why setting the perimeter line width to 0.2mm nets no path there, and only once the machine nozzle width was set below 0.347mm

With the machine nozzle width set to 0.4mm, and both line widths set to 0.1mm, this is the result I get. It fits 4 passes on the one side, and still attempts zero on the other:

[image]

CCS86, 2017-01-27:
On a side note, I just realized that top/bottom line width does not stay linked to line or infill line width. I never had that setting visible. Does it apply to both infill and wall during top/bottom layers? 

I would vote to remove that setting, and just use wall and infill widths globally through the print.

CCS86, 2017-01-27:
[image]

[image]

BagelOrb, 2017-01-27:
If you set the line widths to 0.2mm then you wouldn't be able to fit the outer wall on both sides of the narrow region, because that's smaller than 2x0.2mm.

The outer wall is what's red, the inner walls are whats green, the skin is what's yellow and the infill is what is orange-yellow. Walls on top/bottom layers are still walls. In fact - every layer can have top/bottom skin areas when your object is angled. All prints always have walls on each layer.

CCS86, 2017-01-27:
@BagelOrb , Did you see my last post?

To me, it still seems highly likely that the machine nozzle size setting is leveraged somewhere in the engine.

BagelOrb, 2017-01-29:
The line width settings automatically inherit their value from the nozzle size.

You should also check the Outer Wall Offset setting. I think it is automatically set to the difference between the nozzle size and the outer wall line width.

CCS86, 2017-02-07:
I see the "outer wall inset" setting and think you are right, that it was counteracting the line width setting in a way and preventing the thin path. 

I see the purpose of it, but am curious to test how necessary it is for dimensional accuracy when "under driving" the nozzle size. At least I can take control if need be.

Thanks