Ultimaker/CuraEngine #445: Fan speed issue

Crazydoub, 2017-01-26:
Hi dear friends.
I've got a little problem with cura engin used with repieter 1.6.2.
When i set
Max fan speed ad (height 2mm) _ translated from frenc so not exactly good text_
and Min fan speed to 35, generated gcode is not good.
Last M106 Gcode is this:
```
;LAYER:12
M106 S76
```
which give a speed set to 30% and not 100%
"Max" speed is set at the good position but max value is not good, it should be:
```
;LAYER:12
M106 S256
```
Did i miss something or is it a bug?
Thanks.

Crazydoub, 2017-01-26:
Waiting for source fix, i wrote a little script for repetier post processing:

`# coding: utf-8
#!/usr/bin/python
import sys,re,os, sys, Tkinter, tkFileDialog, tkMessageBox,time, tkSimpleDialog,re,unicodedata

infile = sys.argv[1]
outfile = sys.argv[2]

#infile=r"C:\Users\Crazydoub\AppData\Local\RepetierHost\filter.gcode"

#outfile=r"C:\Users\Crazydoub\AppData\Local\RepetierHost\composition.gcode"

infile = open(infile, "r+")

CONTENU=infile.readlines()

for i in range (len(CONTENU)):

    if 'M106 S' in CONTENU[i]:
        FAN_SPEED=CONTENU[i].replace('M106 S','')
        FAN_SPEED=round(int(FAN_SPEED)*3.368)
        CONTENU[i]='M106 S'+str(FAN_SPEED)+"\n"

outfile = open(outfile, "w+")

for i in range (len(CONTENU)):
        outfile.write(CONTENU[i])

`

Ghostkeeper, 2017-01-26:
There's quite a few settings that the engine uses to control fan speed, all of them in the "Cooling" category. I don't know how RepetierHost names these settings, especially in translated French, but I'll explain with how Cura names them.

I expect that what you're seeing is the "Initial Fan Speed" setting (key `cool_fan_speed_0`). This makes the fan start at 0% and then gradually increase layer per layer towards "Regular Fan Speed" (slightly misleading key `cool_fan_speed_min`). It reaches this regular speed at the layer indicated by "Regular Fan Speed at Layer" (key `cool_fan_full_layer`).

Could you verify if my expectations are met? If it is, then above the first few layers it should be at the regular fan speed, which you say is 35% (`S89`).

There is also the "Maximum Fan Speed" setting (key `cool_fan_speed_max`). This is only reached however when the estimated time spent on the layer is less than some threshold determined by "Regular/Maximum Fan Speed Threshold".

And that's about all the fan speed control that exists in CuraEngine.

Crazydoub, 2017-01-26:
Initial fan speed and max fan speed seetings doesn't exist!
Only max speed at layer...
Cura engine works with this. Fan speed increase at each new layer.
But max value sent by default in gcode is 74 instead of 256.
M106 S76.
It would be good that an ini file would be added to engine (exe+dll+ini) so we could set some settings like fan max speed.
My python script for repetier postprocessing just multiply fan speed values to increase each value and set max to 256....

Crazydoub, 2017-01-26:
Repetier host Regular (defaut)  fan speed is ignored by cura.

Ghostkeeper, 2017-01-27:
It seems that the version of CuraEngine used by RepetierHost is like 2 years old and 3.000 commits behind: https://github.com/repetier/CuraEngine

Crazydoub, 2017-01-27:
So repetier's cura-engine is not your work?

Ghostkeeper, 2017-01-27:
It was branched off of our work 2 years ago (just about when I started working at Ultimaker). They made 4 modifications to it since then. We made 3166 modifications.

Crazydoub, 2017-01-27:
I sent this to repetier's owner:

Hi, there's an issue with curaengine slicer:
Fan speed is not controlled normally!
Criss
I tried to contact Curaengine author which told me that repetier uses a too old version or curaengine!

Here is my post on Github
Thank you for this fast answer!
Initial fan speed and max fan speed seetings doesn't exist!
Only max speed at layer... and Fan speed by default.
Cura engine works with this. Fan speed increase at each new layer.
But max value sent by default in gcode is 74 instead of 256.
M106 S76.
My python script for repetier postprocessing just multiply fan speed values to increase each value and set max to 256....

Here are the g-code samples. Max fan speed is set at 2mm.
We are a lot of people with this issue on the net  

**G-CODE GENERATED BY CURA ENGINE**
G1 F6600 X137.074 Y137.428 E3036.25128
;LAYER:19
M106 S76
G0 F9600 X136.637 Y137.354 Z2.100
;TYPE:WALL-INNER
**G-CODE GENERATED BY CURA ENGINE AND POST-PROCESSED**
G1 F6600 X137.074 Y137.428 E3036.25128
;LAYER:19
M106 S256.0
G0 F9600 X136.637 Y137.354 Z2.100
;TYPE:WALL-INNER
Criss
Could you please share an updated version of cura engine for repetier.
Thanks a lot.

Crazydoub, 2017-01-27:
Ok, thanks, thought that the work still was in your hands!
Difficult to contact them! I'll do my best. Thanks for answer but i think they won't do anything and i'll mus use my postprocessing script...

Ghostkeeper, 2017-01-27:
Yeah, possibly. I suggest you limit it to 255 instead of 256, by the way. I don't know what the printer would do when it encounters 256. It might have a byte overflow and end up at 0% fan.

Crazydoub, 2017-01-27:
Yes, good idea! 256 has been tested but we never know :D
One more time thanks to you and your great job ;p
I'm going to find a good alternative to repetier.

closed 2017-01-27 by Ghostkeeper