Ultimaker/CuraEngine #448: CuraEngine debugging

ghamrawyk, 2017-01-31:
Hi guys, 

I have a beginner's question. I am using ubuntu 16.04 and cmake. I built, ran CuraEngine successfully on a test stl file. Now I want to deeply understand how it works so I was wondering if there is a way to debug it "line by line"? 

I tried to use _gdb_ but didn't work, it gave this error: 

Reading symbols from CuraEngine...(no debugging symbols found)...done.
(gdb) b main
Breakpoint 1 at 0x6525d9
(gdb) run
Starting program: /home/kg/CuraEngine/CuraEngine 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, 0x00000000006525d9 in main ()
(gdb) s
Single stepping until exit from function main,
which has no line number information.
setpriority () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) s

Any suggestions on how to debug line by line? 
Thank you very much,

awhiemstra, 2017-01-31:
You should make sure to build CuraEngine with debugging symbols, otherwise GDB has no idea where it is in the code. You can do that by adding -DCMAKE_BUILD_TYPE=Debug to your CMake command line, then rebuilding. Note that this will make the engine a lot slower, but at the same time what is executed should match a lot more closely with the actual code.

ghamrawyk, 2017-01-31:
Thank you very much for your quick answer. I'll give it a go now and let you know. However I don't understand the last bit, what do you mean by :"what is executed should match a lot more closely with the actual code" ? 
And if I want to run without debugging symbols later on, I should simply rebuild CuraEngine without -DCMAKE_BUILD_TYPE=Debug?

awhiemstra, 2017-01-31:
> However I don't understand the last bit, what do you mean by :"what is executed should match a lot more closely with the actual code" ?

When you compile a project in "release" mode, the compiler is allowed to optimise the code. For example:

```cpp
for(int i = 0; i < 10; ++i)
{
  if(i < 0)
    break;

 doSomething(i);
}
```

The compiler can deduce several things from the code. For starters, since it knows `i` will go from 0 to 10, the `if(i < 0)` condition will never be triggered. It can thus remove that entire check. Furthermore, since the loop has a fixed range, it can unroll it, effectively changing it to read:

```cpp
doSomething(0);
doSomething(1);
doSomething(2);
doSomething(3);
doSomething(4);
doSomething(5);
doSomething(6);
doSomething(7);
doSomething(8);
doSomething(9);
```

This completely removes the need to keep track of a variable (i), decreasing memory usage and increasing speed. While this is a bit of a contrived example, the compiler can actually do a lot more, especially with aggressive optimisations enabled (-O3 in GCC). However, as shown, the actual structure of the code being executed changes, thus making it harder to debug. That is why most of the time a "debug" build will disable optimisations, so what is being executed matches the code your wrote. (Mostly - there are still a few things the compiler does like expand preprocessor macros.)

> And if I want to run without debugging symbols later on, I should simply rebuild CuraEngine without -DCMAKE_BUILD_TYPE=Debug?

Yes.

ghamrawyk, 2017-01-31:
I really appreciate the explanation, things are clear now. Thank you very much. I rebuilt it and now the debugging works. 

Cheers,

BagelOrb, 2017-01-31:
I usually run the debugger straight from my IDE, because I think it has a way better interface, even though it's at the cost of some functionality which gdb does have.

I use KDevelop 4, where you just hit Shift+F9 to start in debugging mode.

ghamrawyk, 2017-02-01:
Perfect! Thank you, I will try KDevelop today. Is there any advice on how to attack CuraEngine in order to deeply understand it? It is a huge code and I am not an expert in programming. Any hints?

Ghostkeeper, 2017-02-01:
There is no one place of overview. The way I see it, the data follows the following general path:
* slicer.cpp (turns 3D models into layers)
* FffPolygonGenerator.cpp (makes insets and computes areas that need to be filled and such)
* FffGcodeWriter.cpp (makes paths)
* gcodeExport.cpp (converts paths to g-code strings)

These files make some calls to various functions all over the place to process the components they need. Reading those four should give a pretty good overview of what the engine does in what order.

ghamrawyk, 2017-02-01:
Thank you very much for the clarification. I think this would be a good way to start, i'll check out those four first. 

Cheers,

closed 2017-02-01 by ghamrawyk