Ultimaker/CuraEngine #450: Certain settings can't be overridden

aswimmingbison, 2017-02-01:
I am trying to create a custom json file for my printer and I cannot get infill_sparse_density to override nor can I get the top_bottom_thickness to override. When I run the command it says that neither one of those are recognised but I copy and pasted them right out of the fdmprinter file. Any help is greatly appreciated. 

Here is the json file http://pastebin.com/jzPuBSsx

Ghostkeeper, 2017-02-02:
Instead of "default_value", override the "value" property.

The reason is a bit technical. The definition files override each other on a per-property basis. I.e. the "description" property overrides the "description" property, the "default_value" property overrides the "default_value" property, and so on. But additionally the "value" property overrides the "default_value" property (the "value" property can be any Python expression while the "default_value" property can only be a JSON value). So if a setting has a "value" property and you override only the "default_value" property, your new "default_value" will eventually get overridden again by the original "value" property.

closed 2017-02-02 by BagelOrb

aswimmingbison, 2017-02-02:
[WARNING] Trying to override unknown setting top_bottom_thickness.
[WARNING] Trying to override unknown setting infill_sparse_density.

This is the warning that I am getting it's not that it is being overridden again but that it doesn't recognise that the settings exist in at all.

Ghostkeeper, 2017-02-06:
Strange. The engine gives these warnings when it finds a setting that it doesn't have a default for. It uses the fdmprinter.def.json document to find these defaults.

You don't happen to be using a very old version of CuraEngine, do you?

aswimmingbison, 2017-02-07:
No I am not. I installed from a version two weeks ago.

BagelOrb, 2017-02-08:
Those are parent settings which are only used to determine the value of
their children.

These settings aren't used by the engine, so the engine doesn't load their
defaults.
However, the frontend doesn't care about this fact and just sends the
non-engine settings as well.

The engine doesn't use top_bottom_thickness, but only uses top_thickness
and bottom_thickness.

BagelOrb, 2017-02-10:
Distinguishing between unregistered settings and non-engine settings is a feature which hasn't been implemented.

chintanjosh, 2017-06-07:
@aswimmingbison @Ghostkeeper Same is the issue which I am facing, with same setting i.e infill_sparse_density.

I am also confused as to how to override the default settings. Can you please guide me ?

Thanks.

aswimmingbison, 2017-06-08:
What you have to do is go down to the lowest possible child setting or as BagelOrb stated you have to use top_thickness and bottom_thickness instead of top_bottom_thickness.

Ghostkeeper, 2017-06-10:
CuraEngine only ever uses settings that have no children in fdmprinter.def.json. `infill_sparse_density` has the child setting `infill_line_distance`. This is the actual distance between the lines that is used to draw infill. The front-end computes the value for `infill_line_distance` based on `infill_sparse_density` and `infill_line_width` such that the desired infill density % is reached. So if you're calling CuraEngine without front-end, you need to provide the correct line distance rather than the density, since only the line distance is used.

Mind that this is not a concern if you're using Cura's front-end, since that calculates the correct value for `infill_line_distance` automatically. Only if you're calling CuraEngine directly from a command line.

chintanjosh, 2017-06-12:
@Ghostkeeper  Thanks for clearing the doubts, will keep that in mind.

Fsmash, 2018-08-29:
I know this has been closed for a while but how do I send settings to CuraEngine then if I'm trying to set a setting with children? Do I calculate all the children values and send that to CuraEngine? Also do I calculate the value of all the other settings that reference the parent setting in the value?

BagelOrb, 2018-08-29:
CuraEngine doesn't use any parent setting values and it doesn't do any
calculations for you, It just uses the default_value from the def.jsopn
files.

So yeah, you must do those calculations yourself, or let the Cura frontend
do it.

Fsmash, 2018-08-30:
Thanks for the reply. I also notice that literally all the settings are passed. Is there some way to stop that? Like some settings have an "enabled" attribute that have conditions on whether or not they are, well, enabled.

If those conditions aren't met is there someway for me to tell CuraEngine not to even get the default value defined in the json? Or does it not even matter that everything is passed and if the conditions aren't met the setting is ignored (or doesn't take precedent over other settings)?

BagelOrb, 2018-08-30:
By design a setting is disabled if and only if the engine doesnt use it -
the setting default value is then ignored.

Fsmash, 2018-08-31:
Alright, that makes sense. Thanks again. You've really helped me out. Just looking at all these settings though and how the reference each other in their value calculations I can't imagine CuraEngine being really easy to use through the command line if you want to change a couple settings. 

I wrote another program to calculate these values (since a lot of settings reference each other) and call CuraEngine. As I understand it, Cura doesn't pass settings to CuraEngine the way we do through the command line, but through sockets (?). Is there any documentation for me to do it this way. And would it be reasonable for me to set up a server to send slice commands to CuraEngine on said server?

BagelOrb, 2018-08-31:
Calling on @ghostkeeper to provide you with some info for this.

Ghostkeeper, 2018-08-31:
Yeah, CuraEngine normally gets it slice commands via a local socket. But we didn't document this well enough in order to make your own implementation. I suggest you take a good look at the [CuraEngineBackend](https://github.com/Ultimaker/Cura/tree/master/plugins/CuraEngineBackend) plug-in in Cura's front-end, which makes the connection from the other side. This connection is established using the [Arcus library](https://github.com/Ultimaker/libArcus), so you'll also need that, and if you compiled CuraEngine yourself you need to make sure that it's compiled with Arcus (which is enabled by default luckily).

There's a couple of caveats to connecting to CuraEngine with a socket connection, because it was designed to interface with Cura's front-end without keeping alternative implementations in mind:
* CuraEngine will send you layer view data. You might want to ignore those messages.
* CuraEngine will send the g-code header separately afterwards. This is because it can only know the printing time and material estimates after slicing has completed. If you slice using the command line, the printing time and material usage in the g-code headers will be placeholder numbers.
* It's possible to send multiple slice commands to CuraEngine and keep the application running. However we discourage that because we found that when you do that, CuraEngine will send different g-code back the second time. Something in CuraEngine is not reset properly when you do this, so it's advisable to restart CuraEngine for every slice in order to get consistent g-code back.

Fsmash, 2018-09-04:
Alright, you guys have been a great help. Thanks again. One last question and I'll try to figure out the rest on my own. The order in which the settings don't matter right? Cura passes the settings through a local socket but also generages cura.log contain '-s' settings passed to CuraEngine. I just noticed from working backwards from this that the order in which the settings are passed are different for each file sliced. But consistent if it's the same file sliced again.

BagelOrb, 2018-09-05:
It only matters to which object settings are passed (global, one-at-a-time,
extruder, per-object).

Otherwise the order doesn't matter.

Fsmash, 2018-09-05:
Ah, the -g and -e options. I see, ok thanks.

In regards to my other question "Also why are some settings included twice in cura.log?", I can only surmise that it first sets a bunch of global settings for calibration then after "-g -e0" starts setting printer settings to override the global settings for the print. Is this correct?

BagelOrb, 2018-09-05:
Ne, the -e switches it to extruder settings. Those are the extruder
settings, - not printer overrides.

Fsmash, 2018-09-13:
@Ghostkeeper 

Sorry to bug you again but you seem like the person to ask this. 

I'm having quite an awful time calculating all these values and when I compare to cura.log it seems as if some values don't even follow the "value" attribute as they're defined in the json files. 

Can you point me to the part of the the Cura front end that accepts and calculates all these settings. I just want to set certain settings and have all the other settings affected calculated for me. 

Also, all the '-s' settings in the cura.log. If I were to send those exact same settings to CuraEngine I should get the same gcode output right? Or am I mistaken?

nallath, 2018-09-14:
The value property can be overridden by profiles again (even changing the formula that is used to calculate it in the first place!). To make it even more complicated; profiles are stacked on top of each other, making it possible to override a value _multiple_ times. 

Right now, this system is tightly coupled with the front end. We were planning on refactoring that in the near future, but this has been pushed a bit back due to some changes in planning from our side.

Fsmash, 2018-09-21:
@nallath 
Thanks for the reply, I figured as much that was what was happening. It would be a real pain to track that all down. Can you tell me exactly when the value attribute will be evaluated? Meaning when would the frontend not just take the default value? Is it when there's a setting whose value has been changed and that affects another setting?

Ghostkeeper, 2018-11-20:
If there is a "value" property, then that always overrides the "default_value" property. If there is a "resolve" property, that always overrides the "value" property (the first time it is evaluated).

bbo-git, 2025-01-30:
I'm wondering, can a setting in the json definitions be made 'not-skipped' by removing it's childrens sections?