Ultimaker/CuraEngine #459: Repetier flavor - M204 used for setting acceleration instead of M201 and M202.

FlameRunner, 2017-02-10:
If I understand the code correctly, no matter which flavour is selected, acceleration-setting g-code is generated by this line: [](https://github.com/Ultimaker/CuraEngine/blob/c7e621eeb5b56dce6fbccaf6d9c46628d3142c80/src/gcodeExport.cpp#L882)
`*output_stream << "M204 S" << PrecisionedDouble{0, acceleration} << new_line; // Print and Travel acceleration`

So, something like: M204 S2000 and that works fine with Marlin.

Problem is, Repetier firmware can't interpret that correctly.
In Repetier, M204 command sets PID parameters (fortunately, the Sxxx argument is omitted).
Setting acceleration in Repetier is done by:
M201 - Set max acceleration in units/s^2 for print moves (M201 X1000 Y1000)
M202 - Set max acceleration in units/s^2 for travel moves (M202 X1000 Y1000)
(source: [](https://github.com/repetier/Repetier-Firmware/wiki/G-codes) )

BagelOrb, 2017-02-10:
We accept pull requests with code which fixes the problem.

I think there is no reason for Ultimaker to invest in fixing such problems.

FlameRunner, 2017-02-10:
Unfortunately, I can't really code and my ability to use GitHub is also limited.

I guess something like this, although being rather crude way, would do the job:

```
if (getFlavor() == EGCodeFlavor::REPETIER)
{
	*output_stream << "M201 X" << PrecisionedDouble{0, acceleration} << " Y" << PrecisionedDouble{0, acceleration} << new_line; // Print acceleration
	*output_stream << "M202 X" << PrecisionedDouble{0, acceleration} << " Y" << PrecisionedDouble{0, acceleration} << new_line; // Travel acceleration
}
else
{
	*output_stream << "M204 S" << PrecisionedDouble{0, acceleration} << new_line; // Print and Travel acceleration
}

current_acceleration = acceleration;
estimateCalculator.setAcceleration(acceleration);
```

It would be much nicer if that code checked if the next move is going to be travel or print and output only one command, but I have absolutely no idea how to do this.

You are probably right that Ultimaker has no reason to fix this, but maybe someone else using Repetier would be willing to pick it up from here?

smartavionics, 2017-02-11:
Hopefully, someone who uses Repetier can test the PR and decide whether it works or not.

brickZA, 2017-04-05:
As far as I can tell it works in Cura 2.5 Beta. Seems to generate M201 / M202 with Repetier flavour and my Wanhao duplicator i3 shows the correct settings on its LCD interface while printing a file sliced with acceleration control.

smartavionics, 2017-04-06:
Thanks for testing this and posting a report.

smartavionics, 2017-04-06:
@FlameRunner if you're happy with this being fixed, you can close this issue.

closed 2017-04-06 by BagelOrb