Ultimaker/CuraEngine #463: First layer after support speed

CCS86, 2017-02-14:
What do you guys think about handling print areas which are the first layer on top of support, with the 1st layer speed? 

I could see it potentially sticking to the support interface more. But I can also see it being able to achieve better surface finish on that skin surface.

BagelOrb, 2017-02-15:
This would be quite nice. Also for bridging and very steep overhangs in
general.

FlameRunner, 2017-02-16:
I think that the area-that-is-right-above-the-support speed should be an independent setting.
For some materials, it would be nice to increase support adhesion by lowering the printing speed. For some, it would be nice to make the support easier to remove by increasing the speed.
I used to do something like this in Slic3r and it proved to be quite effective way of tuning peelable support.

CCS86, 2017-02-23:
I like that idea too

BagelOrb, 2017-02-24:
It's pretty tricky to get that right. If you have different line widths for
overhang, for bridges and for normal skin, then the lines would get
disconnected, leading to ugly bottom surfaces.

You could of course keep the line widths the same, but that would be at the
users discretion.

It should be noted that both these features increase the slice time quite
significantly.

CCS86, 2017-02-28:
@BagelOrb , I missed where line widths were discussed. I think purely a speed setting would allow much faster "recovery", and therefore, better surface finish and strength. I would gladly pay a slicing penalty for it.

BagelOrb, 2017-02-28:
@CCS86 line widths naturally popped into my head when discussing the
intruduction of a new print feature (like skin, infill, walls, etc.)

I am hoping to have functionality like this in Cura 2.5, but I can't
promise anything.

Ghostkeeper, 2017-06-05:
Nope, and not in 2.7 either. Our product manager just closed this for being inactive for 3 months.

CCS86, 2017-06-06:
Well crap.