Ultimaker/CuraEngine #464: Speed & cooling settings for bridging?

JBKingdon, 2017-02-15:
Hi,

I was wondering if options for setting a specific speed to use while bridging and also a fan speed were something that were being considered? Or has this already been discussed and felt to be unnecessary?

Regards,
James

BagelOrb, 2017-02-15:
It has been considered, but it is quite difficult when you really think
about it. Note that part of a skin area might be bridge, while the rest is
normal skin, within one layer.

JBKingdon, 2017-02-15:
But you must already have to identify such areas in order to use a "free air" extrusion volume based on the nozzle diameter. Given that you have to change extrusion for bridging I'm surprised that it's technically difficult to add a fan speed command or pick a different head speed for that section, although I could imagine it's a lot of work to add options to control those settings.

JBKingdon, 2017-02-15:
Ah, so a look at the generated gcode and a search through the codebase suggests that CuraEngine doesn't currently use a free-air extrusion for bridging and just trucks straight across the bridges with the normal perimeter or fill extrusion settings. That came as a surprise, but would certainly make it harder to implement bridging specific cooling and speed settings. This probably explains why I have trouble with bridging with Cura compared to other slicers.

JBKingdon, 2017-02-16:
I did a quick test to confirm what I thought I was seeing in the code. I'm suggesting that the ideal bridging extrusion rate is based on the area of a circle with the size of the nozzle diameter, and if I pick a layer height and extrusion width that is equivalent to the nozzle cross sectional area (CSA) I get the expected reasonable bridging as seen in these first two photos:
[image]
[image]
As per my first post, I'd argue that it is useful to be able to set a specific cooling fan speed and head speed during bridging, and you can see that the longer bridge was better than the first after I manually added some cooling fan.

On the other hand, if the layer height and width gives a cross-sectional area much less than the nozzle CSA then bridging becomes difficult. This test with 0.1mm height, 0.5mm width with a 0.5mm nozzle effectively didn't bridge at all. I stopped it after several passes over the first bridge left only a single strand (which would originally have been straight, I took the photo after having removed it from the plate).

[image]

At the other extreme, if the layer height & width combination results in an area larger than the nozzle CSA you get the expected very droopy bridging

[image]
[image]

I think it would be a really useful enhancement to add a new path type for bridging that could have specific speed and cooling settings, along with a new bridging path generator that uses the nozzle CSA (times a correction factor setting) to generate bridges.

I hope this is useful, and not an irritant!

Regards,
James

Ghostkeeper, 2017-02-17:
You've made some great tests. A lot of that has overlap with this issue though: https://github.com/Ultimaker/CuraEngine/issues/50

Real bridging is part of the "top 50 bugs" we've selected. It has some priority but I can't say for which release we're going to have it implemented.