Ultimaker/CuraEngine #467: How does CuraEngine got slice layer parts with ovrelape polygons?

zidong404, 2017-02-22:
Hello, I am Reading the source code.Now I know the how to Make 'LayerParts' by closed polygongs, My question is:If some of the polygons which created by Slicer() are overlaped by each other,how does the Clipperlib generate correct 'LayerParts'.The code is  `clipper.Execute(ClipperLib::ctUnion, resultPolyTree);`that means the default fill type is even-odd,fallowing is my test:
[image]
    The left is my model, as you can see the bottom suppport parts are overlaped.I've got the firs layer's slice polygons, and the right-up is the result generate by Clipper with none-zero fill type, the right-down is even-odd.Obviously 'none-zero' filltype is correct in this test.
    I am looking for the codes that handling overlape polygons  in curaengine, but I didnt find it,so I was confused how dese curaengine fix this. Follwing is the layer view of the model's first layer in Cura, and Cura makes the correct layerpart.
[image]
     Waiting for answers ,thanks!

BagelOrb, 2017-02-22:
Cura frontend has a setting Union All under the mesh fixes category, which
determines which fill the is used.

<https://cloud.githubusercontent.com/assets/22258646/23218129/5b1246fa-f956-11e6-92ba-63304b8684f3.jpg>
<https://cloud.githubusercontent.com/assets/22258646/23219628/93131a26-f95a-11e6-8804-5aa4aac6c729.jpg>

zidong404, 2017-02-23:
Thanks! A model is loaded into a mesh or split into a meshgroup? Where the function "determines which fill the is used"  be implemented.Could you give me some clues.

BagelOrb, 2017-02-23:
Meshgroups are what's printed in one-at-a-time mode. When you print
normally there's only one meshgroup.

Overlapping meshes are subtracted from each other, while overlapping
volumes within a single mesh are unioned.

The subtraction is done in a function called carveMultipleVolume or
something similar with the word 'carve' in it.

zidong404, 2017-02-23:
Thanks a lot,BagelOrb，I've got it.

closed 2017-02-23 by zidong404