Ultimaker/CuraEngine #469: What command line setting should I use to select Profile=High Quality as shown in Cura GUI

terryspitz, 2017-02-26:
I'm using the following command line on Windows 10:
`"C:\Program Files\Cura 2.4\CuraEngine.exe" slice -j "c:\Program Files\Cura 2.4\resources\definitions\prusa_i3_mk2.def.json" -l Marvin.stl -o Marvin.gcode`

-j selects the Printer, how should i select Material and Profile.

BagelOrb, 2017-02-26:
There is no such option. You can only pass settings to the engine with "-s
setting=value".
Note that when you pass the json to the engine it uses the 'default_value'
rather than the 'value' function.
The engine doesn't compute any of the interdependencies between the
Settings.

terryspitz, 2017-02-26:
So the Material and Profile are just GUI features?  Is there some way to save the command line with settings from the GUI?

BagelOrb, 2017-02-26:
The GUI spits out a command line call in the log, which is close to what a
command line call would be when you call the engine directly.
I don't know for sure what was missing from the logged command. I think it
was something like setting extruder_nr=0 or 1 depending on the extruder.

BagelOrb, 2017-07-23:
[answered]

closed 2017-07-23 by BagelOrb