Ultimaker/CuraEngine #471: fill_density value setting

jackyting, 2017-02-27:
What is the parameter to set the fill density? I have to find the corresponding infill_sparse_density and infill_line_distance two parameters in the JSON configuration file, I set and change the value of infill_sparse_density times, but when the printer found no effect. Set the value of infill_line_distance, found that only infill_line_distance=0 when the model becomes hollow, and other values are very sparse kind of filling effect. For example, I would like to set a model for the filling density of 50% or set to the value of the 0% should be set when the value is more appropriate?（under the command line environment）

BagelOrb, 2017-02-27:
For the command line the fill density is not used. It is only a setting which the frontend uses to determine the infill line distance..

Setting the infil line distance should work.

jackyting, 2017-02-28:
okay，I set up a comprehensive infill_line_distance, infill_overlap and infill_overlap_mm value, fill the effect.

closed 2017-03-01 by jackyting