Ultimaker/CuraEngine #477: Variable layer height

Gregwar, 2017-03-09:
Just referencing this as a possible futute feature, I think the idea is quite reasonable since the layer height makes more sense to be small depending on which slice is involved
http://www.prusaprinters.org/smooth-variable-layer-height-awesome-supports-slic3r-prusa-edition/

BagelOrb, 2017-03-09:
> awesome-supports-slic3r-prusa-edition/

Yeah idea has been around for a long while.

However, it's quite difficult to implement in CuraEngine.
Moreover, a different layer height requires a lot of different settings to
pront well. Making all those settings depend on layer height in the right
way is rather difficult.

It is on my wish list, though.

Gregwar, 2017-03-09:
I guess the hardest is to define how the part geometry influences the generated layers height
After that, the settings are indeed important but it can be some kind of range (finest vs roughest), so it is understandable by the user

BagelOrb, 2017-03-09:
I think it's hard to convert all other algorithms which depend on layer
height.

The hardest part is to make all settings depend on the layer height,
though. All setting values need to be converted into formula's.

JustAnother1, 2017-03-10:
In the Prusa video he manually tunes the layer hight. I can change the Layer hight in Cura and then just push print with the changed settings, right?

So wouldn't it be just a change in the part that comes up with the Layers? That part would need some more intelligence sorting out the correct heights for each layer. But after that wouldn't it be like "here another layer with x hight" and the rest would just do it's thing?

I'm probably missing something here. But what?

BagelOrb, 2017-03-11:
Top/bottom skin, support overhang angle, ooze shield, all 3d infill
patterns, support interface and infill combine count are all dependent on
layer height. There's probably even more!

Ghostkeeper, 2017-03-13:
One difficulty is to refactor the engine to make sure it can handle different layer heights for different layers. That's the difficulty you mention but it's probably not very hard indeed.

But there's also a more inherent difficulty: The layer height has a lot of influence on other settings. For instance, support overhang angle needs to be lower if you have thicker layers. So when the user wants to have an angle of 50 degrees that works for 0.1mm layers, but if the algorithm decides that some piece gets a layer height of 0.3mm then that layer will get terrible overhang.

JustAnother1, 2017-03-15:
It probably makes sense to split this into two features:
1.) enabling the engine to have a layer hight associated with each layer. Refactoring the Layer Hight to not be a global but a layer local variable. Then all calculations will be done in the layer with the associated hight. Once this works the next step would be to add a mechanism to set the different layer heights. The Prusa video shows manual setting of layer heights. If the wrong hight setting causes bad overhangs, then that is the users fault.

2.) automatic detection of the best suited layer hight. That was not shown in the Prusa video. And that would be a more complicated feature. This probably has to happen in a very early stage. Here the similarities between layers (->thicker layers) or overhangs and details (-> thinner layers) need to be looked at to define the best hight for the the layers.

I see 2) as a feature that will take some time. It also makes no sense to start with 2) before 1) works. And maybe we find out that humans are just better in deciding on the best hight,...
But that should be no argument to not start with the refactoring/architectural changes to make 1) happen.

daid, 2017-03-15:
Actually, I think 1 is the hard part. 2 isn't even that difficult.

Gregwar, 2017-03-15:
For 2, I'd say:

- Substract the layer with the previous one (polygon operation w/ clipper)
- Offsets negative the obtained polygon to "dilate it"
- Measure the area

The layer height could be related to this overhang area (that would be the dynamic layer parameter)

Gregwar, 2017-03-15:
(Dilatation may make the difference between a really slight sphere and a big overhang)

BagelOrb, 2017-03-15:
You don't have polygons to apply clipper operations on before you determine
at which height to slice that layer.

Gregwar, 2017-03-15:
Thats correct, but you can optimize the layer height iteratively until the area is smallest than a given value

I confess that this would require a lot of call to the slicing method

BagelOrb, 2017-03-15:
I can implement it a lot better. The problem is really having separate
layer heights per layer.

For example: when computing the overhang of a layer, do you use the layer
height of that layer, or the supporting layer below?

There are loads of problems like that...

Gregwar, 2017-03-15:
I'd say the height of the layer below because overhang is about estimating the empty space below right ?

Gregwar, 2017-03-15:
I understand that software engineering and taking this in account is really complicated  though, fidling with the auto-adaptation is likely the fun part :)

BagelOrb, 2017-03-15:
The fun part:

Create a graph for the minimal face angle at each height.

Convert angle values to layer height values.

'Flatten' to get rid of too sudden layer height changes.

Sample layer heights. Make a 45 degree angle line from the point in the
graph at the height of the last layer and at zero layer height and see
where it intersects the graph.

I understand that software engineering and taking this in account is really
complicated though, fidling with the auto-adaptation is likely the fun part
:)

BagelOrb, 2017-03-15:
The real fun part comes when you do more interesting statistics than just
the minimum angle of each face at a given height.
Stuff like the average angle or the weighted average by how much the face
is present on the layer.

That's where you can get creative ^^

Gregwar, 2017-03-15:
The face angle is indeed a good approach, I agree

I wonder how much you'd need to flatten. I mean how much we can print a 0.2 layer and then a 0.07 or absolutely not 

The "Convert angle values to layer height values." is not the easy part, imagine you have the top of a sphere, it is clear that you want finest layer height, but if you have a stair, you may not need it. However, this can be discussed because that would give you better accuracy on where the steps of the stairs are.

Also, a surface does not correspond to ONE height but to an interval of height, which may be more complicated to handle (maybe having a discrete array and compute the smallest angle every 10 microns?)

Gregwar, 2017-03-15:
(Illustrating)
[image]

The "flatten" here is actually a limit on the derivative, with always keeping the lowest values

BagelOrb, 2017-03-15:
Thanks for the illustration; that's exactly what i meant!

We may want to limit at which angle we include a face...

Gregwar, 2017-03-15:
@BagelOrb I don't know, the top-of-the-sphere example (that is what is done in the article) suggests that fine layers are useful even with small angles don't you think?

BagelOrb, 2017-03-15:
Hmm yeah. So maybe we should preprocess the graph and remove spikes smaller
than the lowest layer height, before we 'flatten' it.

Gregwar, 2017-03-16:
We can also argue that having finest layers may be a good point in the "stair" case also, because it offers a better z accuracy, so maybe this would be the expected behaviour

JustAnother1, 2017-03-16:
@daid I fear that there might be unsolvable situation in 2) where an automatic solution will always fail in one model or the other. 1) on the other hand is a software refactoring. So it is a lot of work, and as pointed out some thought should be given on features that cover also neighboring layers to which hight to use. But I think besides that it is a well understood endeavor.
You can split 1 in smaller steps:

a) create the layer specific hight variable. (Initialized to the layer hight -> all layers have the same hight)

b) switch all usages to the layer specific variable.

 Here some code locations need further thought to figure out which layer hight is the right to use on features that need to take the layer below or above into account. So maybe b1 is just switching the obvious reads of layer hight. b2) is then solving the different things.

c) once all these issues of b) have been resolved : remove the global layer hight variable

d) add code to initialize the layer variables to different values.

To implement this feature this would be a possible migration plan. The steps are relatively small and can be taken without breaking too much. But a progress could be seen while moving from a) to b1),..

Gregwar, 2017-03-16:
@JustAnother1 I note that there is already a special case for the very first layer, that may also help during the refactoring. That means that at some points there would be a layer height map object that would contain an height for every layer instead of two parameters (this is a lot of work that need to be done by someone that knows well this part of the code I guess).

JustAnother1, 2017-03-17:
@Gregwar  You are right. This special case for the first layer could go away. 

I agree that this suggestion is a lot of work for someone who knows the code base very well. That is also the reason that I do not offer a pull request (I don't know the code) but only these comments.

mikko70, 2017-11-07:
When this happen, i hope possibility raise bed and nozzle temperature about 5 celsius for first layers. If you haven't warming bed, cold bed need raise temp first.

ChrisTerBeke, 2017-11-08:
Started WIP pull request for this feature: https://github.com/Ultimaker/CuraEngine/pull/647.

closed 2018-01-29 by ChrisTerBeke