Ultimaker/CuraEngine #478: OMP gcode writing badness

smartavionics, 2017-03-13:
Hi Tim, I commented on your closed PR and I don't know if you saw that so I am creating a new issue.

I'm seeing some bad gcode appearing after the OMP gcode writing branch was merged

Gcode before omp merge...

[image]

Gcode after omp merge

[image]

It looks like some extrusion moves start or end at the wrong z?

BagelOrb, 2017-03-13:
That doesn't look too good!

Could you attach your curaproject.3mf and your gcode, please?

smartavionics, 2017-03-13:
I can get the problem showing up simply by turning off combing. Let me know if you need the model but it seems not to be model specific.

Here's another example...

[image]

smartavionics, 2017-03-14:
I don't know if it's the same problem but with combing enabled I still see some weirdness that isn't there when not using the omp code.

In this example, look at the two perimeter lines of the large circular hole and the inset0 line of the smaller hole inside, they are all kinked at one end.

[image]

BagelOrb, 2017-03-14:
The reason I am asking for the gcode is that I can then verify whether this is an actual gcode problem or a problem of your gcode viewer.

I cannot reproduce the problems when looking in RepetierHost.

Please attach your .curaproject.3mf and your gcode.

Please also try to find the problem in the gcode itself, so that you can verify the problem is actually present in the gcode.

BagelOrb, 2017-03-14:
Ah I can reproduce the problem now.
The problem occurs when you have retraction hop enabled.

This is why it's good to attach your project file: it includes all your settings.

smartavionics, 2017-03-14:
I can't save the project. It fails with an empty message box in the GUI and the following on stderr...

ERROR - LocalFileOutputDevice.LocalFileOutputDevicePlugin.requestWrite [93]: There are no file types available to write with!

Here's the model and the gcode. I just used the UM3 machine default profile with combing disabled.

[UM3_bracket_bigheli.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/841048/UM3_bracket_bigheli.gcode.txt)
[bracket_bigheli.stl.txt](https://github.com/Ultimaker/CuraEngine/files/841053/bracket_bigheli.stl.txt)

BagelOrb, 2017-03-14:
Ok thanks. I was testing on the UM2.... :P

BagelOrb, 2017-03-14:
Fixed it. I should have fixed it ASAP, but ah well...

Thanks for reporting this!

BagelOrb, 2017-03-14:
I cannot reproduce the weird kinks at the wall of the inside hole. I think is a visualization error.

closed 2017-03-14 by BagelOrb

smartavionics, 2017-03-14:
The weird kinks have also gone away now.

BagelOrb, 2017-03-14:
Ok good!

BagelOrb, 2017-03-14:
About the project saving bug: we are not able to reproduce your problem. It's very mysterious why you get that error. Are you sure you are up to date with all 2.5 branches of all repositories involved?

If so: could you maybe attach your log file?

BagelOrb, 2017-03-14:
Perhaps it's better to make a new issue in https://github.com/Ultimaker/Cura for that..

smartavionics, 2017-03-14:
It was because I didn't have the libsavitar stuff installed. OK, now.

BagelOrb, 2017-03-14:
Ah ok.