Ultimaker/CuraEngine #482: How to output metadata such as calculated print volume using CuraEngine

Quintox303, 2017-03-18:
It appears as though the only metadata that is contained in the gcode output is the print time. How does cura GUI know this information if it is not included in the output of CuraEngine? Or do I need to look somewhere else to output this information

BagelOrb, 2017-03-18:
There are several ways in which the egi communicates such information.

The engine communicates with the frontend via an internal socket, via a
library called Arcus.

The gcode contains some of the data in the gcode header, but that depends
on the type of gcode you are generating.

The engine outputs some information over stderr. The command line option
'-v' makes the engine output more stuff.

Quintox303, 2017-03-18:
Thanks for that. I noticed after running with -v there was a line at the bottom saying "Filament: 13976" for example. Could you tell me what unit this is measured in? I'm assuming thats mm of extruded filament

BagelOrb, 2017-03-18:
Either that or cubic mm. I'm not sure. Might even depend on the type of
gcode you are generating.

Quintox303, 2017-03-18:
What do you mean by the type of gcode you are generating? What are the types

BagelOrb, 2017-03-18:
Reprap, marlin volumetric, ultigcode, Repetier, Griffin. Each firmware type
requires a slightly different gcode spec.

Ghostkeeper, 2017-03-21:
In the RepRap flavour it says:

    ;Filament used: 0.110303m

The Griffin flavour says:

    ;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:705

The UltiGCode flavour says:

    ;MATERIAL:740

BagelOrb, 2017-07-23:
[answered]

closed 2017-07-23 by BagelOrb

devangsaini123, 2018-04-17:
Not able to get the model-size/bounding box even with the verbose flag (-v). Also, I am trying capture the surface area of the geometry. Is there an option to tweak the output metadata according to my requirements.

BagelOrb, 2018-04-17:
There is no option of CuraEngine to output those..

devangsaini123, 2018-04-17:
Thanks for the prompt reply!!
In such a scenario, how does the Cura GUI display the dimensions like - ( 55.0 x 79.3 x 44.8 mm). Do I need to analyze the generated g-code separately for capturing these dimensions?

BagelOrb, 2018-04-24:
Those dimensions are not the dimensions of the gcode, but of the input
files!

Ghostkeeper, 2018-05-11:
Note that the Griffin g-code flavour puts the final print dimensions in the header. This also includes stuff like the prime position so that throws the print dimensions completely out of whack.