Ultimaker/CuraEngine #489: Short prime tower required for multi-extruder spiralized models

smartavionics, 2017-04-02:
I've been working on improving the print quality when spiralizing models that use multiple extruders. Some progress has been made but the big issue at the moment is that if the prime tower is not enabled, after the nozzle switch it does the post switch prime directly onto the object being printed and this causes a huge blob.
Enabling the prime tower should get over that but that breaks the spiralization because it has to build the tower the same height as the model and so on every layer goes off and works on the tower.
I am wondering if it would be possible to have a prime tower that only has a layer for each nozzle switch and when the switch occurs the nozzle z has to be reduced down to the height of the tower and then increased again back up to the height of the model. Is this possible? If so, can you give a rough idea of what is required to be done?

The most obvious problem would be ensuring clearance for tall models, for some models it would be impossible to print the low tower without striking the model with some part of the extruder/carriage.

Thanks.

smartavionics, 2017-04-02:
Just had another thought, Tim. How about having a "virtual prime tower" that gets used only for spiralization.
So the nozzle gets moved to the prime tower position and when it is to be used again it does the big prime extrude (into thin air) and then moves back to the print. OK, so the prime gets squirted into air and makes a mess but that won't make a blob will it? It will just produce a thin extrude that won't hardly stick to anything and can just be brushed away afterwards. And if there are not many extruder swaps, there won't be many of those pieces of extrude anyway.
Whaddyathink? Is it worth me pursuing that idea?

smartavionics, 2017-04-02:
I'm working on a PR that does the prime into thin air so I think we can ignore this issue, closing it.

closed 2017-04-02 by smartavionics

BagelOrb, 2017-04-04:
Yeah a prime tower would kind of break the whole idea of spiralize, because you'd need to print the prime tower every layer until the switch of filament.

What printer are you using?
If you are using an Ultimaker 3 then all extruders which are used throughout the print are primed in the first layer, so the blob should never occur mid print.
If not then there is a bug elsewhere in the process...

smartavionics, 2017-04-04:
Hi Tim,
Unfortunately, I don't have a dual extruder printer so someone else is doing the testing on a UM3. There's a thread on the forum where I have been posting gcode for them to try. If you are interested, take a look at the last one I posted (UM3_SpiralMultiMaterial_5). Also, my PR is up to date with the current code so you can see what I am doing (ah I see you have just commented on that)...