Ultimaker/CuraEngine #490: Agressive zig-zag on sloped shells

larsch, 2017-04-02:
When slicing sloped shells, e.g. at 45 degrees, CuraEngine generates an aggressive toolpath of little zig-zags with many travel moves on the inner shell wall. It causes shaking and vibrations even on UM style printers.

The zig-zag is visible when using transparent filaments.

With low wall count (1 or 2), the hotend tends to soften the existing overhanging shell lines and drag them into the part and "up".

Changing wall thickness, wall line count, or infill density does not change this.

Perhaps simplifying to a single extrusion line would make the result cleaner and reduce vibrations.

[image]

STL File: [slope-cura.zip](https://github.com/Ultimaker/CuraEngine/files/888273/slope-cura.zip)

Previous discussions: 
- https://ultimaker.com/en/community/10538-cura-adding-evil-fill-causing-warts-ruining-prints
- https://ultimaker.com/en/community/22115-zigzag-extrusion-between-wall-and-infill

smartavionics, 2017-04-02:
Hi, in Cura 2.5 with have a feature that lets you expand skins that go into infill. So in your example above you could ask it to expand the skin by a few mm and although that would make the skin more visible it would reduce the frequency (lines are longer so it takes longer to do) so should stop your printer falling apart!
We have also added an option to shrink skins before they are expanded which stops narrow skins being expanded so you want to set the shrink distance to 0 otherwise the narrow skin won't be expanded.
In my original implementation it was possible to shrink the narrow skin completely so that it went away but that  wasn't approved so in 2.5 you can't make the skin smaller than the original area (personally, I think that is a shame and it would solve your problem.)

larsch, 2017-04-03:
I'd much prefer just rounding the x-y-plane shell width to an integer number of linear passes.

smartavionics, 2017-04-03:
I think I understand what you're saying. Add extra perimeters on those edges that slope inwards rather than having a narrow band of skin. Still, when you get 2.5 give the above feature a go and it may help.

Ghostkeeper, 2017-04-03:
Setting the top/bottom pattern to concentric also fixes the problem, but it produces worse overhangs at the top.

BagelOrb, 2017-07-23:
The Skin Reference Wall issue makes it possible to avoid skin being generated at sloped surfaces.

wawanbreton, 2024-04-05:
The described behavior is obsolete, no relevant in newer versions, closing.

closed 2024-04-05 by wawanbreton