Ultimaker/CuraEngine #494: Random prime tower on work

sanderux, 2017-04-07:
Cura 2.5.1 on linux (appimage) decided to put a prime tower on top of a small support structure consistently over 3 prints. Id did it at approximately layer 3. I cannot share the stl unfortunately. 

Printer: UM3
Material: PC
Extruder: 2
Supports enabled using extruder 2 (PC)
buildplate adhesion: skirt, also tested raft

Currently testing windows with 2.4

BagelOrb, 2017-04-07:
It's hard to understand what you mean from written language alone. Could
you attach a screenshot of the problem area in layer view??

Ghostkeeper, 2017-04-07:
It should put the prime tower inside the little grey square on the build plate (you can see it before printing). You shouldn't be able to put anything inside the grey square.

Were you able to position objects on top of the grey square, or did it produce the prime tower outside of the grey square?

sanderux, 2017-04-07:
No it was outside of the grey area but it wasnt really a prime tower as it did not change z axis. I saw it stop all movement and flow material fast for about 5 seconds. The amount extruded was sinilar to a prime tower.

I will try to look if i see something in the layers. I can confirm this doesnt happen on 2.4

sanderux, 2017-04-07:
see the issue here in 3 aborted prints:
https://goo.gl/photos/CYXq9utHi4rvxMLS9

Ghostkeeper, 2017-04-07:
Maybe it's the normal prime? It should draw a small circle in the positions somewhere down below on the right side of the bed where you're also not allowed to print (and again not allow you to place an object there).

sanderux, 2017-04-07:
Ok, however;
- The real prime tower was already placed.
- the object is outside of the gray non printable area.
- there was no material change so no reason for it to start doing a prime tower
- it also happens when i disable prime towers for dual extrusion

I will rotate the object so this bit is no longer located in the bottom right area

sanderux, 2017-04-07:
The layers show no sign of this behavior but as the nozzle isnt moving it's hard to see.
[image]

sanderux, 2017-04-07:
I modified the model to no longer need support material in this location.
Now it did the same thing on top of different support material several layers later
I witnessed it doing this a second time on the same location just now

[image]

BagelOrb, 2017-04-08:
This might be related to an issue which caused not all of the gcode to be sent to the printer.

Please retest by printing from USB.

Ghostkeeper, 2017-04-10:
Or with the second beta version. It should've been fixed there, right?

BagelOrb, 2017-07-23:
Needed more info. Issue is outdated.

closed 2017-07-23 by BagelOrb

sanderux, 2017-07-23:
It has not happened again since 2.6