Ultimaker/CuraEngine #496: run curaEngine on server with command line, But Export an empty gcode file.

MickeyLeez, 2017-04-10:
However, it works correctly on the local host by using "CuraEngine slice -j xxx.def.json -o xxx.gcode -l xxx.stl." in a java project. And I had double-checked the paths json file and stl file. (The server environment was windows server 2008, while the loca host was windows 7)

BagelOrb, 2017-04-10:
If you add `-v` then you will get more output.

It might be the case that your model is below the build plate. Try `-s
center_object=True`

MickeyLeez, 2017-04-10:
Thanks for your advice.
What does the '-s center_object=True' mean ? And it still can not explain why it just worked correctly on local host.
Now I am going to add "-v" and "-s center_object=True".

Ghostkeeper, 2017-04-10:
`-s center_object=True` will cause the model to be centred on the build plate. If it's below the build plate, that would move it to be on top.

BagelOrb, 2017-07-23:
Not enough info. Issue is outdated.

closed 2017-07-23 by BagelOrb