Ultimaker/CuraEngine #497: Question: why does writePathWithCoasting() use addLastCoastedVolume() ?

smartavionics, 2017-04-12:
Calling addLastCoastedVolume() adds the volume that wasn't extruded during the coast to the extruder's prime_volume so that next time the extruder un-retracts it will extrude the coasted volume. This seems wrong to me because it will cause extra plastic (blob) to appear at the start of the next extrude. Why would you want that to occur?

BagelOrb, 2017-04-12:
The idea was to not loose any e-steps because of coasting, but in the end
it didn't really work and we have up on coasting.

It would be better to do it like such: coasting causes the last segment of
a line to be underextruded and the first segment of the next line to be
overextruded.

That way coasting bleeds off and builds up the extruder pressure required
for the current print speed.

I don't see any reason for coasting which only involves the bleeding off
part.

I never got around to implementing the buildup.

smartavionics, 2017-04-12:
Understood. Thanks for the speedy reply. Closing...

closed 2017-04-12 by smartavionics