Ultimaker/CuraEngine #500: Test suite: StringTest fails

thopiekar, 2017-04-18:
Is that a known problem? 
Also: Is there a way to make the test verbose? I guess the following output is not that useful:
```
dh_auto_test || true
	make -j4 test ARGS\+=-j4
make[2]: Entering directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
Running tests...
/usr/bin/ctest --force-new-ctest-process -j4
Test project /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu
    Start 1: SparseGridTest
    Start 2: LinearAlg2DTest
    Start 3: PolygonUtilsTest
    Start 4: PolygonTest
1/5 Test #2: LinearAlg2DTest ..................   Passed    0.01 sec
2/5 Test #3: PolygonUtilsTest .................   Passed    0.01 sec
3/5 Test #4: PolygonTest ......................   Passed    0.00 sec
4/5 Test #1: SparseGridTest ...................   Passed    0.01 sec
    Start 5: StringTest
5/5 Test #5: StringTest .......................***Failed    0.01 sec
.............F.................

80% tests passed, 1 tests failed out of 5

Total Test time (real) =   0.02 sec

The following tests FAILED:
	  5 - StringTest (Failed)
Errors while running CTest
Makefile:119: recipe for target 'test' failed
make[2]: *** [test] Error 8
make[2]: Leaving directory '/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu'
dh_auto_test: make -j4 test ARGS+=-j4 returned exit code 2
make[1]: Leaving directory '/<<PKGBUILDDIR>>'
```

Ghostkeeper, 2017-04-19:
Yeah you can run `ctest --verbose` in the build directory.

Ghostkeeper, 2017-04-19:
Oh wait, this is the Engine test suite, nevermind then. It's not possible as far as I know because the executables don't produce output.

Ghostkeeper, 2017-04-19:
<FailedTests>
  	<FailedTest id="13">
  	  <Name>cura::StringTest::writeInt2mmTestMax</Name>
  	  <FailureType>Assertion</FailureType>
  	  <Location>
  	    <File>/home/ruben/Projects/CuraEngine/tests/utils/StringTest.cpp</File>
  	    <Line>97</Line>
  	  </Location>
  	  <Message>assertion failed
    - Expression: in == out
    - The integer 9214157878975800 was printed as &apos;1330109.752&apos; which was interpreted as 1330109752 rather than 9214157878975800!
      </Message>
  	</FailedTest>
    </FailedTests>

BagelOrb, 2017-04-19:
Ah yes. Apparently the `writeIntmm` fails on very large numbers. I don;t think there's a great risk in this, though... Numbers shouldn't really get that high any way.

BagelOrb, 2017-07-23:
I cannot reproduce this.

closed 2017-07-23 by BagelOrb

Ghostkeeper, 2017-08-03:
That's because you modified the test such that it doesn't test the specific case any more that wasn't working as expected: https://github.com/Ultimaker/CuraEngine/commit/1f1c2c686d740c79e23c91aacee82b25347e571e