Ultimaker/CuraEngine #506: How to export settings from Cura to run with CuraEngine?

ChristopherHoffman, 2017-04-26:
How would I go about setting up a profile using Cura 2.X (currently 2.5), and then exporting that profile to run against CuraEngine.exe directly? I'm able to slice using the command line by defining .Json profiles, but the only options to export directly from Cura are to save as a .curaprofile, which I cannot find a way to load it into CuraEngine.

Do I have to write a program to convert each setting in the .curaprofile to a -s command line switch, or is there an easier way to work directly with a .curaprofile file? The same does with materials (.xml.fdm_material) and machine settings (no export?)

BagelOrb, 2017-04-30:
You could check the output of the engine when you slice with a given profile. It should contain a line which is close to what the engine call should be. It's not perfect, though. I know there is some stuff you need to manually fix, but I can't remember what.

closed 2017-04-30 by BagelOrb

Ghostkeeper, 2017-05-04:
The alternative is to write your own script. This needs to:
* Apply the custom profile on top of the quality profile.
* Apply the result on top of the correct subset of the material.
* Apply the result on top of the current variant.
* Apply the result on top of the current printer.
* Apply the result on top of FDM printer (or whatever the inheritance structure is of your printer).
* Convert this to a string of `-s x=y` parameters for the command line.

ChristopherHoffman, 2017-05-04:
Thanks for the responses! I tried the Cura's logged command line switches, but unfortunately it seems that trying to run it through node's child_process gives an error about the command line argument being too long with all those switches. I also see that Cura is using a socket connection to CuraEngine, as opposed to just calling CuraEngine.exe slice -s..., does using the socket connection allow you to load -s switches one at a time, which might allow me to bypass that length restriction? I will also have to test out how to pass through g-code start/stop scripts, since they have line breaks.

Another thought to decrease the number of command line arguments, would I be able to load in the .def.json for the printer (which inherits from FDMprinter.def.json and FDM extruder, etc), and then write a script to create the switches from the material and curaprofile files? Its good to know the ordering that should be applied, thank you @Ghostkeeper , but would CuraEngine do the layering correctly if I apply both a -j (for the .def.json) and -s arguments afterwards?

Ghostkeeper, 2017-05-05:
CuraEngine doesn't apply all of the profiles, only FDMPrinter and other JSON files (the 4th and 5th of the items in my previous comment).

Cura's front end indeed uses a socket connection. It doesn't send via -s but it actually sends a Protobuf message over the socket which contains all settings. It therefore has no (practical) limitation of the amount of settings it can override.

SirCmpwn, 2018-02-11:
It would be nice if there was an option to export a script from the GUI rather than actually performing the slicing.

alekseisasin, 2018-02-12:
@SirCmpwn , what do you mean under "export a script". If I am correct, you mean: save all properties which are required for slicing a specific model via console (call CuraEngine without GUI). I am right ?

SirCmpwn, 2018-02-12:
@alekseisasin yep

BagelOrb, 2018-02-12:
Yeah it already does that. There are some bugs with command line slicing tho. The transformation matrix should be passed before loading an object rather than after.

Good luck!

SirCmpwn, 2018-02-12:
Are you referring to the command line dumped to the log? I attempted to use this to automate some prints but the results were different from when I used the cura GUI.

Ghostkeeper, 2018-03-07:
The command line dumped in the log doesn't include the transformations made on the model. The front-end by default puts the centre of the model as coordinate 0,0,0 and then (if drop to buildplate is enabled) aligns the bottom side of the model to the build plate. The back-end doesn't do all that. But there's the `mesh_position_x`, `mesh_position_y` and `mesh_position_z` settings to align it manually via the command line.