Ultimaker/CuraEngine #507: Need to have option to disable inter-layer smoothing in new spiralization code - discuss

smartavionics, 2017-04-30:
For models with nicely curved surfaces (i.e. typical smooth vase), the new spiralization code does a good job of hiding the z-seam. It does this by smoothing the x/y coodinates  across layers so at the point where one layer ends and the next starts it doesn't introduce an ugly step.
Where it is definitely less good is for models that have very fine details in the z-direction as these tend to get "smeared" by the smoothing process.
I am suggesting that we add a checkbox setting to disable the inter-layer smoothing so that the user can choose between smoothed layers with (hopefully) no visible z-seam but with possible degradation of the surface detail or non-smoothed layers with potentially a visible z-seam but surface detail intact.
I am quite happy to implement this if given the thumbs up.

Ghostkeeper, 2017-05-05:
Fine by me. This approaches what used to be Surface Mode (before it broke), except that spiralize still produces bottom skin.

smartavionics, 2017-05-09:
This has now been incorporated, thanks.

closed 2017-05-09 by smartavionics