Ultimaker/CuraEngine #513: How to determine the support position in programming

RNDAlex, 2017-05-08:
In fact, I had a few more questions about support when I was reading CuraEngine code. Can someone give me some answers or hints?
1. Like the title, how to determine where the support area should be? All in detectOverhangPoints? but it cant be used unless the tag "support_use_towers" has been setted.
2. I dont know what is "support_angle" meaning exactly. the tangent of angle would be used by CuraEngine but in Cura 2.3 you can set the value to 90! The input value was be filtered by frontend, right? BTW, I am not sure the following expression is right. That makes me confused. (not cotangent but tangent?) 
`int max_dist_from_lower_layer = tanAngle * supportLayerThickness;`
3. How to understand the difference operation in double-for-loop?
`for (Polygons& poly_here : overhang_points_here)
{
         for (const Polygons& poly_below : overhang_points_below)
          {
                 poly_here = poly_here.difference(poly_below.offset(supportMinAreaSqrt*2));
          }
 }`
how to "// make sure we have the lowest point"?
4. Can I design my own support-pattern? support_mesh would be helpfully? (need some hints)

BagelOrb, 2017-05-08:
1 detectOverhangPoints is for underhand - model geometry which is not
supported whatsoever, while overhang is supported but the angle is too
horizontal.

The whole class in support.cpp is concerned with which areas there should
be support.

2 support_angle is the overhang angle. See the description when you hover
over the setting in the frontend.

3 that piece of code is just the difference between to vectors of polygons.
It's part of the underhand detection, though.

4 if you want your own support pattern than you should make a new infill
pattern. The way in which the support areas are filled in is done in
FffGcodeWriter.

RNDAlex, 2017-05-08:
@BagelOrb Thanks for ur reply.

closed 2017-05-10 by RNDAlex