Ultimaker/CuraEngine #514: Whats the method "processInfillMesh" for

RNDAlex, 2017-05-10:
Yes, I had read the comment above the function declaration, but I still not sure how to be an infill mesh.
Can someone show me the principle about that in graphics?

BagelOrb, 2017-05-10:
Just try it out in the frontend.
Add a couple of meshes.
Select one and in the per-object settings enable Infill Mesh.
Give the infill mesh a different infill percentage in the per-object settings.
Make the infill mesh overlap with half of the other meshes.

RNDAlex, 2017-05-11:
And before do that,  I should disable the setting "Ensure models are kept apart" ...
Ok, thanks for your suggestion. Next time I will practise in the frontend first if I met something confusing.
@BagelOrb

closed 2017-05-11 by RNDAlex

Ghostkeeper, 2017-05-12:
Infill meshes themselves are not printed.

The part of a (normal) model that overlaps with an infill mesh will get printed with the settings of that infill mesh. This includes the infill density, but also the extruder, speeds, etc. Very powerful feature.