Ultimaker/CuraEngine #515: setting "support angle" about the cone

RNDAlex, 2017-05-11:
1. It seems like the polygons no changed/not to do anything when "joined = joined.offset(supportJoinDistance).offset(-supportJoinDistance)".
2. old question:
[image]
The value of Support Overhang Angle was set to 19 and the cone surrounded by support structure.
[image]
Then changed the value to 20 all support dispeared. idk why

BagelOrb, 2017-05-11:
1. that line does a close operation. See for example:
http://homepages.inf.ed.ac.uk/rbf/HIPR2/close.htm
2. That's because the cone has an angle of 20 degrees

RNDAlex, 2017-05-12:
got it，thanks for your help. @BagelOrb

closed 2017-05-12 by RNDAlex