Ultimaker/CuraEngine #517: Problems on layer change - Very long moves before.

povlhp, 2017-05-16:
Not sure if this is the place, or if it should be a cura issue.
[2xgcode.zip](https://github.com/Ultimaker/CuraEngine/files/1005692/2xgcode.zip)

I am trying to print the 3dlabprint wing parts, and on Cura 2.5 I get issues on every print. Seems like it is not extruding/underextruding on each layer start. Cura 2.3.1 works fine and gives perfect output. Same STL, same settings. This is parts with 0.4mm wall thickness, 0.2mm layer height.

There is difference in the code generated. The working one does not have a long move (and retraction) before starting. The Cura 2.5 code does have a long move with retraction. Here is one sample of layer change, Y is changed from 107 to 18. So a very long travel. Should be better picking a nearby spot and start from there. Maybe this far a travel will starve the nozzle of filament ? Or the pulled back filament will be cooled too much down, taking time to start flowing again ?

I assume I could do something by printer settings - Less retraction and get more stringing ?. And Cura 2.5 code is estimated to be slower and use more filament. So not likely an improvement.

[CODE]
G1 X79.774 Y107.81 E557.15665
G0 F5400 X79.708 Y107.621
G1 F3600 E555.95665
G0 F5400 X79.078 Y18.44
;TIME_ELAPSED:413.080342
;LAYER:18
G0 X79.078 Y18.44 Z3.9
;TYPE:WALL-OUTER
G1 F3600 E557.21901
[/CODE]

2x code included. Code does seem to print. But the front wall is more or less completely missing in all its 5+mm width. So it is a pretty far distance I do not get any filament extruded. Higher layers when that space gets narrower, I also have extrusion issues around the corner. For some reason the corner is sharp all the way up. Only issue before and after.

povlhp, 2017-05-16:
Did disable the retraction.Still same problem (in a slightly different place). Now the stringing instead of printing moved around the corner. 
I still have the long wasted movement before layer change. Not sure about the 2 moves before layer change. The 2.5 used engine seems to always start near same place. The older engine picks a close spot, so every layer starts in a different place. This is good for strength.

`G1 X79.493 Y107.81 E246.99542
G1 X79.774 Y107.81 E247.00179
G0 F5400 X79.708 Y107.621
G0 X79.078 Y18.44
;TIME_ELAPSED:168.941334
;LAYER:18
G0 X79.078 Y18.44 Z3.9
;TYPE:WALL-OUTER
G1 F3600 X80.393 Y18.415 E247.05209
G1 X81.997 Y18.4 E247.11345
`

povlhp, 2017-05-16:
Printing on a MeCreator2.  Bed 160x160mm. 
Problem appear very early / from beginning. No solid endwall at lowest y position. And when it narrows the extrusion issues moves around the corner. 2cm is plenty to see error. 

When printing 2 elements at same time I think I had same issue on travels. But dropped multiple elements and went to 2.3.1. I had thin wall sections ending with a blob. Like filament was in a narrower line than 0.4mm and then extruder caught up on direction change. 

Other issue I experienced in vase mode was z axis going up and down. Destroying things printed when moving down to print more.

BagelOrb, 2017-05-16:
Sounds like you have coasting enabled. Try disabling that setting.

BagelOrb, 2017-05-16:
About the layer starting position: there's no way to change that in Cura
2.5 and later versions.

povlhp, 2017-05-16:
Coasting NOT enabled. For me, 2.5 is just not possible to use for 3dlabprint planes.
Might be my printer/extruder, as the code do look fine. Problem seems to be triggered by the long move (with or without retraction).

povlhp, 2017-05-17:
Here are photos of some of my earlier prints. Both from 2.5, both with multiple elements. separated like 20-30mm.

One has holes in it. And as you can see it is for many layers in the same place. Could be extraction. The other photo, I added extra extrusion after retraction. The result is I get a blob when that flows out. But then a pretty thin line after that.

I have tried gcode.ws, looking if I get less extrusion, but it will not color my lines when I select mm filament per mm move. But it looks like it goes from 0.011mm to 0.040mm when I look at the legend. All walls should be the same width (0.4mm). Structure is open, and no infill. Just thin walls. Does anybody have a working tool ? Or should I do something in Perl myself (yes, I am that old :-) )

It could be hardware related. Using Mk8 extruder. That after long move, a bit of the pressure buildup has oozd out - But I do not have much stringing (when no retract), or when retract that the nozzle can keep up with heating the retracted filament, so basicly it will do the blob of stuff in nozzle, and then can't keep up with heating what else is feeded in.  But still, I need to look at extrusion per mm move I think.

BTW: I am running 115% extrusion. Extruder moves 3-4% too little when measured on its own.  And there is no noticeable problem in Cura 2.3.1 (could be random starting place distributing/hiding the error). As for start position, shouldn't it be possible to pick a random spot for each layer ? I know it would take a few CPU cycles to move say  thread number * 2 segments forward compared to the threads last ending position, and then start there. But it would be cheap, at least with few threads (typically 4-16 I guess ?)

[image]
[image]

povlhp, 2017-05-17:
Analyzing the gcode around the layer change.

Here is the layer change. we have low e/mm on the short distances.
If E is the stepper motor, then 1mm E = 2.41 mm2 area
For 1mm movement we need 1mm*0.4*0.2 = 0.08mm3, so we need 1mm extrusion per 30mm print, or around 0.03mm per mm.

It seems the code is generating the right amount of extrusion.  The first 4 line segments is giving me the problem in the real world.

One thing, could it be that the extruder is not completing its retraction recovery in time ? Will older Marlin wait for one command to complete before starting the next ? 

Should I try something like G4 (dwell) after all G1 with no X/Y ? Giving time to re-heat after retraction/reload ? Might be an alternative to extra-extrusion after retract. And maybe the right one.

I am a 3D printing n00b, so just trying wild troubleshooting guesses.

G1 F3600 E555.95665
  Extruded: -1.20000000000005
G0 F5400 X79.078 Y18.44
  Move: 4294967.29599999
;TIME_ELAPSED:413.080342
;LAYER:18
G0 X79.078 Y18.44 Z3.9
  Move: 4294967.29599999
;TYPE:WALL-OUTER
G1 F3600 E557.21901
  Extruded: 1.26236000000006
G1 X80.393 Y18.415 E557.26932
  Print: dist: 1.3152  extr: 0.0503  e/mm: 0.0383
G1 X81.997 Y18.4 E557.33067
  Print: dist: 1.6041  extr: 0.0614  e/mm: 0.0382
G1 X83.851 Y18.414 E557.40159
  Print: dist: 1.8541  extr: 0.0709  e/mm: 0.0383
G1 X85.663 Y18.44 E557.4709
  Print: dist: 1.8122  extr: 0.0693  e/mm: 0.0382
G1 X85.663 Y19.748 E557.52093
  Print: dist: 1.3080  extr: 0.0500  e/mm: 0.0382
G1 X85.663 Y19.812 E557.52338
  Print: dist: 0.0640  extr: 0.0024  e/mm: 0.0383
G1 X85.663 Y20.63 E557.55467
  Print: dist: 0.8180  extr: 0.0313  e/mm: 0.0383
G1 X85.663 Y23.977 E557.68269
  Print: dist: 3.3470  extr: 0.1280  e/mm: 0.0382
G1 X87.634 Y23.978 E557.75808

Ghostkeeper, 2017-05-17:
As for the starting position: The reason that "closest" was removed as an option for the layer starting position is because we implemented multi-threading in 2.5. This means that we can no longer guarantee that the previous layer is computed before the next layer is started and so we don't know the ending position of the previous layer and we can't find the closest position to start the next layer at. One way to solve this would be to compute the line order of a layer as a Hamiltonian Path cycle (multi-threaded) and then cut the cycle at the closest position in sequence.

We do still provide the Z Seam Alignment setting which allows you to distribute the starting position of a piece of the print by selecting the Random option. This doesn't influence the order in which pieces are printed.

> One thing, could it be that the extruder is not completing its retraction recovery in time ? Will older Marlin wait for one command to complete before starting the next ?

It should. We assume that Marlin only does one thing at a time. However I don't think that Marlin will wait for the pressure in the nozzle to build up if you have a very big nozzle.

> Does anybody have a working tool ? [to visualise the g-code]

You can load g-code back into Cura and it should display the line thicknesses and the line types. You can also try Repetier Host which doesn't show the line types in its visualisation but it's a bit more reliable with weird vertical moves or just exception cases in general.

> Should I try something like G4 (dwell) after all G1 with no X/Y ? Giving time to re-heat after retraction/reload ?

This can be achieved with a post-processing script in Cura. It's not something we will likely implement though. I don't think it would help much, but it might do a little good for extrusion. It could also cause blips on your surface though.

povlhp, 2017-05-17:
I am on a standard 0.4mm nozzle, and a standard heating block. Nothing special about my setup, and since I get an extra blob if I do extra extrusion I assume the heater can keep up, and since the issue also occurs without retraction, that is not likely the reason.

Will write a perl script to pause after the ;LAYER and the feed forward, and test again. Easier than to write a plugin. At least for testing.

But the funny thing is, I do not see the problem at all under 2.3.1. But what I see there visualizing the code is, that it always seems to finish a layer with a print. Will try the Z-seam to see if it hides the problem to the level where it disappears.

BagelOrb, 2017-05-17:
Perhaps you have the setting Retract at Layer Change enabled.

How about Cura 2.4, is that version working correctly for you?

povlhp, 2017-05-17:
Changed the Z seam alignment to Random. Now I do not have long idle travels, and I have a less visible issue. Now when I look closely, I see under extrusion / layer separation in different places. So it clearly has to do with starting a new layer.
No retraction on layer change. Retraction minimum travel is 0.8mm.

On the models, I do not see the issue, my Marvin test is close to perfect, and my tiny ship is nice as well. But they all have multiple outer layers.

Still have long travels. 10% of my travels are 50+mm. Oh well, issue must be elsewhere. Will create some test models. Looks like shortest move in 2.3.1 was the setting to work. Also worked in 2.4

Random start point gave me 120-ish moves > 50. Shortest gave me 12. Not sure what move length will give me issues.  If it is retraction - it could be that extrusion after extraction is bad. So my issue was likely different defaults. Will get back when I find out more.

BagelOrb, 2017-07-23:
Did you ever find out more?

saft6luck, 2017-10-08:
Having the same problem with 2.7 here. Once in a while there are thin walls which are not printing (basicaly printing with very thin lines) after some "random" moves before.
Reading the file with Cura 2.7 doesn't show any problem but Marvin is obviously not able to print the whole wall with the good extrusion any more. At next jump it might work like a charm for the rest of the layer until exactly this wall again.

Looks like in the picture above.

Setting 0.4mm nozzle and nearly all options tested.
Tested 2.3 and works well.

tajcovek, 2018-02-06:
I'm experiencing a very similar issue with 3.1 64 bit  & 3.0.4 64 bit. Did anyone solve it yet?

nallath, 2018-02-06:
No, because we never got enough info to fix it.

BagelOrb, 2018-02-06:
Please please upload the workspace file (.curaprojext.3mf) which produces this problem!

tajcovek, 2018-02-06:
I could not find a way to upload it here. So here is a dropbox link
https://www.dropbox.com/sh/vnwzb4zlo9g8qr6/AAD1gjUOEBWW0QU7clbTwiJia?dl=0
[Also here is a link to my reddit post trying to solve this with some images of the issue.](https://www.reddit.com/r/FixMyPrint/comments/7vcaw3/backside_of_circular_prints_look_awful_temp/)

tajcovek, 2018-02-07:
I printed the same part on Cura 2.7.0 and had no issue

cangelis, 2018-02-07:
Related to #679?

povlhp, 2018-02-07:
Now I know a lot more about 3D printing. I am pretty sure that my original issue was down to simple physics.  Retract of course retracts, removing pressure from the nozzle, sucking air in with the retract. When moving forward again, it will take time to build the same pressure.
At the time I was using old Marlin firmware, Jerk of 20, and 1000 acceleration, Maybe I was just moving faster initially than I would be able to melt the filament ? Maybe it was the rough precision of my extruder, or my crappy extruders ability to push colder filament thru (100 e-steps/mm, so 1 step =  0.3mm forward movement (0,2 layers, 0.4 wide)).
Have not printed these parts for a long time, but I would assume I do not have the same issues today with upgraded printer (geared extruder, bowden, more calibration knowledge, both e-steps, and filament multiplier etc).. So likely hardware issue combined with bad user parameters.

tajcovek, 2018-02-07:
Unfortunately for me, this is not a physics issue. No retractions here. Tried on many many different setting temp, flow, reset e-steps. Many test prints later the only solution was removing cura 3+ and reinstalling 2.7

BagelOrb, 2018-02-07:
As of Cura 2.8 or something a layer cannot start where the previous layer
left off. This limitation was implemented on purpose to speed up the
slicing process. Perhaps that is what you are experiencing.

tajcovek, 2018-02-07:
I couldn't tell you. I know I'm experiencing awful results printing anything that is round in cura 3+.
Printing at random "sharpest corner" make all the missed lines line up on an angle at the back of the ring. Printing "random" for z alignment resulted in missed lines randomly around the ring.
some reason its not extruding enough filament at the start of each layer. Its as if coast is on, but its not.
I have not figured a way to compensate for the issue with any settings available. 
I uploaded those file as asked, hope they help solve this issue.

tajcovek, 2018-02-07:
Installed latest version and tried "shortest" for Z-Alignment. That seemed to do the trick. Also tried with coast on and se to 000mm3 and that produced even better results. somewhat odd.

povlhp, 2018-02-08:
Coasting is like retracts. It is leak prevention - ie remove presure from nozzle before travels.

zanney, 2018-05-09:
I am experiencing under extrusion at start of new layers CURA 3... causing very thin wall for 1 to 30mm of new layer. Outside wall. Is there a way to start new layers on inside walls? I'm new to 3d printing and this seems to be an issue that is software related. Thank you and Best Regards, Terry

smartavionics, 2018-05-10:
Hello @zanney , make sure you do not have the option to print outer before inner walls enabled.