Ultimaker/CuraEngine #518: Raft outer-perimeter optimisation

inkychris, 2017-05-20:
## The Problem
As it currently stands (in Cura GUI), there are no options to alter the perimeter for a raft. I typically use a raft only for things that are difficult to print on the first layer (eg. lots of jagged or multiple small polygons). As it currently stands, depending on the "Raft Extra Margin" property, the perimeter of the raft can also end up with some jagged edges or tight corners as seen in figure 1 (which simply moves the issue of first layer failure from the part itself to the raft). The current solution to this is to increase the "Extra Margin" property until these are smoothed out as seen in figure 2; however, this has the downside of increasing print time and filament required.

[image]
[image]

## Potential Solution 1
As a (seemingly) fairly simple solution, I thought it could be useful to add a "Raft Minimum Perimeter Radius" property to avoid situations where the raft perimeter would have tight corners like figure 1 but allow the raft extra margin to remain low to save time and filament. One problem I can see with this is that very large values for Raft Min Perimeter Radius would need to result in automatically increasing the raft extra margin, to maintain the extra margin round sharp convex corners.

## Potential Solution 2
Considering the potential downsides of having a "Raft Minimum Perimeter Radius" property, an alternative solution could be to add "Raft Shape" options. The first would be default/current behaviour that might be described as lasso mode (like in drawing software) or just "None" if that is more backward compatible with config files, and then a "Rectangle" and "Ellipse" option (or perhaps others if needed). The raft outer perimeter would be based on finding the smallest (selected) shape that encloses the parts being sliced and then adding the raft extra margin to it (for the rectangle, also a possible corner radius property if desired). Blue line is the smallest shape that surrounds the part and the pink would be dependent on the Extra Margin property. For the rectangle, the green line represents the effect of the corner radius option:

[image]
[image]

The raft shape options remove the possibility of the raft outer perimeter being unnecessarily complicated depending on the part being sliced, therefore increasing the likelihood of a successful first layer which (for me) is often the reason for using a raft in the first place!

BagelOrb, 2017-05-21:
Your proposed solutions are actually harder than a proper solution.

We could add a smoothing setting which defines the minimum radius of an
inward corner of the raft.
Outward corners are already as smooth as a circle with the radius of the
Raft Extra Margin.

That is - we could do:
`
raft_outline = first_layer_outline.offset(extra_margin +
smoothing).offset(-smoothing);
`

When you make the smoothing setting big the raft ouline will approximate
the convex hull, similar to the skirt.

Note that that implementation also closes small holes in the raft, but I
see that more as a feature than as a bug.

Ghostkeeper, 2017-05-21:
I think implementing a boolean setting to make the raft convex would suffice, rather than a comparatively complex smoothing function.

BagelOrb, 2017-05-22:
@ devs: see CURA-3854

inkychris, 2017-05-22:
The skirt seems to generate an appropriate outline, would it be possible to use the way the skirt outline is generated for the raft? As a side note, the ability to set a number of perimeters could further improve the success rate of the raft (as discussed in [Cura issue 942](https://github.com/Ultimaker/Cura/issues/942)).

BagelOrb, 2017-05-22:
I've already implemented it, but the rest of the Cura team needs to be
involved in the process to decide how and when to include it in the
software.

Hopefully it will be added in Cura 2.7.
2.6 beta is soon to be released.