Ultimaker/CuraEngine #519: Problems running on Mac OS X

fcacarminati, 2017-05-20:
The Cura engine workbench seems not to be supported on the Mac. When creating a new printer I have

`Running the Python command 'createMachineDef' failed:
Traceback (most recent call last):
File "/Applications/FreeCAD 2.app/Contents/Mod/cura_engine/Commands.py", line 51, in Activated
panel = MachineDef.PrintBedTaskPanel()
File "/Applications/FreeCAD 2.app/Contents/Mod/cura_engine/MachineDef.py", line 246, in init
self.form = uic.loadUi(self.homeDir + "/.FreeCAD/Mod/FreeCAD-CuraEngine-Plugin/MachineDef.ui")
File "", line 4, in Cannot open file`

To fix this I had to do 

mkdir ~/.FreeCAD
ln -s /Applications/FreeCAD.app/Contents/Mod/ ~/.FreeCAD/

However even with this mod the program will fail later on. First of all the path has to be set to the Cura engine

/Applications/Cura.app/Contents/MacOS/CuraEngine

but even so things are not working. I have then to do the following

diff FreeCAD-CuraEngine-Plugin/SlicerPanel.py /Applications/FreeCAD.app/Contents/Mod/FreeCAD-CuraEngine-Plugin/SlicerPanel.py
172c172
< docDir = FreeCAD.ActiveDocument.FileName.replace(docName + ".fcstd", "")
---
> docDir = FreeCAD.ActiveDocument.FileName.replace(docName + ".FCStd", "")
257c257,260
< _cmdList.insert(1, "-v")
---
> _cmdList.insert(1, "slice")
> _cmdList.insert(2, "-v")
> _cmdList.insert(3, "-j")
> _cmdList.insert(4, "/Applications/Cura.app//Contents/Resources/resources/definitions/101Hero.def.json")
259a263
> _cmdList.append("-l")
In this way a file is produced, however... this is really a poor-man solution. BTW, I have to add one printer at random. There should be a default file and a pull-down menu

Ghostkeeper, 2017-05-21:
This doesn't look like something we made. I never knew there was a FreeCAD mod for CuraEngine.

Perhaps you should take this over to https://github.com/cblt2l/FreeCAD-CuraEngine-Plugin/issues . Or perhaps @cblt2l would like to take a look at this issue?

fcacarminati, 2017-05-22:
I have done it. Thanks for the suggestion. Best regards

closed 2017-05-22 by fcacarminati