Ultimaker/CuraEngine #521: Unnecessary retractions and movement

warbunnies, 2017-05-23:
I decided to try out Cura because I had heard good things and I wanted to compare it to my experience with s3d and slic3r. However I am having a major issue with the number of retractions and the order of movements being performed. 

Instead of printing a set of perimeters, it will print one loop then retract and jump and start a different set of perimeters, before coming back to the original perimeter set to finish the second one, then retract again and go do something else. The firmware shouldn't be retracting more than once for a set of perimeters. Attached is a photo, I have turned on coasting so that you can see how it is retracting an absurd amount when it does not need to be. 

The version I am using was downloaded today from the ultimaker website. Is there a setting I can change or is this a glitch?

[Unnecessary-retractions.pdf](https://github.com/Ultimaker/CuraEngine/files/1020727/Unnecessary-retractions.pdf)

smartavionics, 2017-05-23:
It just so happens that I am just about to post a PR that attempts to address this issue. It's in it's infancy but looking promising.

warbunnies, 2017-05-23:
Well thats lucky! So has cura always been like that? There are a lot of smart features built in but that seems like something that should have been fixed a long time ago. Either way thanks, I'll definitely test it out.

smartavionics, 2017-05-23:
Hi. I'm a relative newcomer to Cura myself so I don't know if it has always behaved like it does in this regard. To test it you will have to build CuraEngine from the source as I don't know when this will be integrated into the code base.

warbunnies, 2017-05-23:
O... XD I'm working on a dc micro extruder right now so I aint got time to mess around with that. I wish ya luck though!

BagelOrb, 2017-07-23:
Duplicate of https://github.com/Ultimaker/CuraEngine/issues/225

closed 2017-07-23 by BagelOrb