Ultimaker/CuraEngine #522: Difference of two gcode generated by Cura and CuraEngine

RNDAlex, 2017-05-23:
As I know, there are more details and unnecessary parameters in raw code of engine. What makes me confused is that gcode created by engine much larger than the one generated by Cura. I check two gcode and find out some diffs, one of them is WALL-INNER part included in raw one(gcode) but not in Cura one(gcode). That means engine changed a lot in Cura? I want to know how to reduce the code useless in CuraEngine one.BTW in my test, the same options and parameters had been used in two engines.

Ghostkeeper, 2017-05-24:
When you slice in Cura it calls CuraEngine underwater. They should be the same.

However, Cura puts several layers of profiles on top of the defaults. The defaults are in fdmprinter.def.json which is typically provided when slicing using CuraEngine from the command line, but when you use Cura's front-end then you will have a machine added that changes a couple of settings, and usually some quality profiles or materials.

If a profile ends up setting the Wall Line Count to 1, I can understand why you aren't getting any WALL-INNER parts when you slice via Cura's front-end.

RNDAlex, 2017-05-25:
@Ghostkeeper 
That's right! this problem is about the value of "wall_line_count". it had been changed to 1 but default is 2.
Thanks again.

closed 2017-05-25 by RNDAlex

chintanjosh, 2017-06-06:
@RNDAlex @Ghostkeeper , hey guys I am facing the same issue. I am getting two very different gcodes from GUI and from command line.

Ultimaker2 Printer is selected in my Cura GUI  
from the log I found this command being passed 

`2017-06-06 16:31:40,883 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Calling engine with: ['/opt/cura/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/opt/cura/share/cura/resources/definitions/fdmprinter.def.json', '']
`
Which is followed by a various settings.

If I run the following command from the command line 
`./CuraEngine slice -v -j /opt/cura/share/cura/resources/definitions/fdmprinter.def.json -o /home/xyz/Desktop/gcodes/command_line_pause-start_100.gcode -l /home/chintan/Downloads/pigrrl-2-raspberry-pi-game-console-by-adafruit/pause-start.stl`

I get very different results...

but if I pass this command 

`./CuraEngine slice -v -j /opt/cura/share/cura/resources/definitions/ultimaker2.def.json -o /home/chintan/Desktop/gcodes/command_line_pause-start_100.gcode -l /home/chintan/Downloads/pigrrl-2-raspberry-pi-game-console-by-adafruit/pause-start.stl`

the gcode files are quite same, although there are minor variations (2k-3k lines of gcode)

So what could be causing the issue ?

Also am I doing it correctly ? 

I also plan to pass custom settings from command line but I don't know how, is that done by appending
-s to the command or do I need another json file ? 

Thanks

Ghostkeeper, 2017-06-10:
When the GUI initiates a slice, it takes the Ultimaker 2 definition (`ultimaker2.def.json`) along with your preferred quality profile and flattens it to get the raw setting values for CuraEngine. You can see in Cura's log exactly which settings are sent to CuraEngine.

Ultimaker 2 overrides quite a number of settings. That's why you get very different results if you load from `-j fdmprinter.def.json` and not very different from `-j ultimaker2.def.json`. Additionally, the front end overrides some settings with a quality profile. I believe the UM2 profiles override only 8 basic settings or so, like the print speed and layer height. Unfortunately, the engine doesn't understand these profile files, so in order to slice it correctly -- exactly the same -- you'd have to manually enter the settings from your preferred quality profile to the engine command via `-s layer_height=0.15` and such. For instance, the low quality profile can be found here: https://github.com/Ultimaker/Cura/blob/master/resources/quality/low.inst.cfg

Additionally, there are some minor changes when slicing via the command line because we then have to output the g-code sequentially. For instance, the time estimate will always be 666 because when we have to write the time estimate (at the top of the file) we haven't generated the g-code yet.

chintanjosh, 2017-06-12:
@Ghostkeeper
Thank you very much for clearing the doubts,
I was wondering and was about to raise the question regarding the devil's number (666) being there on time and material estimate. Although, we get that values from the command output so I guess that should do for now.

Thanks