Ultimaker/CuraEngine #523: Retraction not generated

stokmanis, 2017-05-23:
With retraction settings enabled, Cura 2.5.0 does not generate the retracting.
Here are the settings
[image]
Here is the STL test file, which have 10mm travel moves
[test_missing_retraction.zip](https://github.com/Ultimaker/CuraEngine/files/1021314/test_missing_retraction.zip)
[image]

Here is the except from g code, which shows that extruder does not retract
G1 X90.6 Y90.6 **E3.75174**
G0 F3600 X90.67 Y90.67
G0 X96.83 Y96.83
G0 X96.9 Y96.9
G1 F1800 X96.9 Y103.1 **E4.06106**

Maybe I am missing some setting :)

smartavionics, 2017-05-23:
Is combing enabled for skins? That could zap the retract.

stokmanis, 2017-05-23:
Yes, I had Combing Mode set to All (default). When I set it to "No Skin", the retraction was in place. Thanks a lot, issue solved.
The "combing" in the first layer does not make sense, as the bottom surface will be messy due to the extra travel garbage laying below the normal layer.

closed 2017-05-23 by stokmanis