Ultimaker/CuraEngine #526: Connecting via command socket

masklu, 2017-05-24:
If I connecting via command socket like this

CuraEngine.exe connect 127.0.0.1:12312 -j fdmprinter.def.json
 welcome

what should I send.

BagelOrb, 2017-05-24:
Maybe @nallath can help you better.

awhiemstra, 2017-05-24:
When using the command socket, we expect a binary stream of data that is provided in a format implemented in the [Arcus] library. Arcus defines a wire format and uses [Protobuf] for serializing the actual data. The Protobuf protocol file is included in this repository (Cura.proto in the root directory).

Note that the socket connection is intended for two-way communication between a front-end application and the engine. It is not intended as a generic server.

[Arcus]: https://github.com/ultimaker/libArcus
[Protobuf]: https://github.com/google/protobuf

BagelOrb, 2017-07-23:
[answered]

closed 2017-07-23 by BagelOrb