Ultimaker/CuraEngine #527: Compilation Error in 'make' command (Branch 2.6)

chintanjosh, 2017-06-01:
Hello,

I took the code of 2.6 branch along with other repo of libArcus, Ultimaker, and Cura with the same branch. I wanted to use 2.5 but that branch was not available with libArcus hence this branch.

I had this issue previously but with the branch of 2.1 with this repo and for libArcus it went away, and my Cura started happily, however it was not able to slice the files (Maybe because Cura was of 2.5 branch), 
anyway I decided that lets have everything for the same branch and hence, I am using 2.6 branch.

So far, I have installed libArcus without any errors.

And after that I went for CuraEngine.

> cmake .. 
this runs successfully

> make
The error is thrown here 

`Scanning dependencies of target clipper
[  1%] Building CXX object CMakeFiles/clipper.dir/libs/clipper/clipper.cpp.o
[  2%] Linking CXX static library libclipper.a
[  2%] Built target clipper
[  4%] Running C++ protocol buffer compiler on Cura.proto
Scanning dependencies of target _CuraEngine
[  5%] Building CXX object CMakeFiles/_CuraEngine.dir/src/bridge.cpp.o
[  7%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
In file included from /home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/Comb.h:11:0,
                 from /home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/LayerPlan.h:8,
                 from /home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/FffGcodeWriter.h:15,
                 from /home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/FffProcessor.h:5,
                 from /home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/commandSocket.cpp:7:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(Args ...)’:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected ‘,’ before ‘...’ token
             [args...]()
                  ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:18: error: expected identifier before ‘...’ token
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: error: parameter packs not expanded with ‘...’:
             [args...]()
                     ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:34:21: note:         ‘args’
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:36:34: error: expansion pattern ‘args’ contains no argument packs
                 return new T(args...);
                                  ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<_Res(_ArgTypes ...)>&, Args ...)’:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:21: error: expected identifier before ‘...’ token
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:52:24: note:         ‘args’
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:54:36: error: expansion pattern ‘args’ contains no argument packs
                 return new T(f(args...));
                                    ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In constructor ‘cura::LazyInitialization<T, Args>::LazyInitialization(const std::function<T*(Args ...)>&, Args ...)’:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected ‘,’ before ‘...’ token
             [f, args...]()
                     ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:21: error: expected identifier before ‘...’ token
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: error: parameter packs not expanded with ‘...’:
             [f, args...]()
                        ^
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:68:24: note:         ‘args’
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h: In lambda function:
/home/xyz/Desktop/cura-folder/2.6/CuraEngine/src/pathPlanning/../utils/LazyInitialization.h:70:30: error: expansion pattern ‘args’ contains no argument packs
                 return f(args...);
                              ^
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2
`

***
Versions:

GNU 4.8
Python 3.5
Protobuf 3.3
Cmake 3.7

***
OS: Ubuntu 14.04

smartavionics, 2017-06-01:
Hi, I don't know what the min required g++ version is to build CuraEngine but I am using version 5.4.0 on Ubuntu 16.04.2 quite happily. That error above looks like lambda function syntax so maybe you need a more recent compiler to cope with that language feature?

chintanjosh, 2017-06-02:
Thanks, with a newer version of GNU(4.9+) everything got started.
However there are now issues of GUI, but would put that in Cura repo.
Thanks for the help.

closed 2017-06-02 by chintanjosh

Ghostkeeper, 2017-06-05:
We indeed require version 4.9.2 or higher.

chintanjosh, 2017-06-06:
@Ghostkeeper  Thanks, that was what I used after the issue. And it got fixed.
Is there any forum where I can discuss my doubts regarding the CuraEngine/Cura 
Or do I have to raise the issue ?

smartavionics, 2017-06-06:
You can post to https://ultimaker.com/en/community/software-and-firmware or raise an issue on github.

chintanjosh, 2017-06-06:
@smartavionics Thanks I'll look into it.