Ultimaker/CuraEngine #528: CuraEngine stops in the middle of slicing process

ynotstartups, 2017-06-01:
Tried on both Window and Ubuntu, same error
Tried on Cura frontend and CuraEngine command line too. same error

2017-06-01 12:57:17,360 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'wall_0_extruder_nr'
2017-06-01 12:57:17,385 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 4294967295. Resetting process and socket.
2017-06-01 12:57:17,854 - INFO - UM.Backend.Backend._onSocketError [196]: Backend crashed or closed.

smartavionics, 2017-06-02:
Hi, yes, I think so. If you are using the master engine code, you also need to keep up to date with the front end and the definition files.

BagelOrb, 2017-06-02:
Yep.

closed 2017-06-02 by BagelOrb