Ultimaker/CuraEngine #530: [2.3.1] How to override settings from the command line without using *.json files ?

chintanjosh, 2017-06-07:
Hello all,
    For overriding default settings(for eg: infill_sparse_density) this is what I am doing :

- Creating a custom_fdmprinter.def.json which is same as fdmprinter.def.json
- Creating a custom_printer_settings.def.json which inherits custom_fdmprinter.def.json

    Now, what if I want to pass the settings from command line, instead of saving it to json file and then passing it. I am passing it this way.

`./CuraEngine slice -v -j /path/to/custom_printer_settings.def.json -o /path/to/output.gcode -l /path/of/file.stl -s over_ride_setting=override_value -s over_ride_setting_1=override_value1`

    But the over_ride_setting is not taking the effect.
    Is there any other way to override the defined settings other then storing it in the json files ? 

> CuraEngine : 2.3.1
> OS: Ubuntu 14.04

BagelOrb, 2017-06-07:
You are using the right way.

I assume `override_value1` is actually the right value..

Do note that only leaf settings can be used by the engine. No setting which
has children settings is ever used by the engine.

chintanjosh, 2017-06-07:
Thanks,
Okay, say for example 'infill' is having multiple children so it can't be used. But, its children are not having any further child and is a leaf setting so that can be used right ?

BagelOrb, 2017-06-07:
Yep that's what I mean.

The parent settings are only used by the frontend for users to manipulate
the children settings more easily. The backend (CuraEngine) doesn't use
them.

chintanjosh, 2017-06-08:
Okay so from the frontend there are no issues, I write the setting here and my job is done smoothly,
but from the backend what are the settings that needs to be passed for the exact same effect, as 
"infill" or "infill_sparse_density" will not work as they are not the leaf nodes. 

Here is the screen shot of the infill frontend setting.
[image]

chintanjosh, 2017-06-08:
I was being naive, all this while my answer was there in front of me, I got the all the values needed from the cura's log.

For any one stumbling on this in future the way to override settings from the command line is 

`./CuraEngine -v -j /path/to/your/file.json -o /path/to/output.gcode -l /path/to/input.stl -s override_setting=value -s ... ` so on and so forth.

Thanks @BagelOrb I now got your point, only leaf node setting gets the cookie. :+1: ;)

Thank you for the help.

closed 2017-06-08 by chintanjosh

yablokoff, 2017-10-09:
@BagelOrb @chintanjosh guys, thank you so much for bringing this up this. Saved me days!
Just one question left – in CURA frontend parent parameters are updated by leaf parameters but rule of update is not completely clear. For example I want users to input infill density but don't know how to convert them from Infill Line Distance. 

Also not clear what parameters are leaves: just Infill Line Distance (which derives from Cura interface) or also Infill Overlap, Top Layers and Bottom Layers.

Thanks for help in advance!

Ghostkeeper, 2017-10-18:
That's two questions :stuck_out_tongue_winking_eye: 

For the first question: The rule of updating those settings differs greatly from setting to setting. Most of the time this rule will be listed in fdmprinter.def.json at the [value](https://github.com/Ultimaker/Cura/blob/276232dae5030cc002232409bf7a31bea7d4a0f5/resources/definitions/fdmprinter.def.json#L1357) property of the setting. Infill line distance happens to be a very complex one because it's dependent on the infill pattern and we have a lot of infill patterns now. Sometimes a profile will overwrite this formula though, especially if you're using the Ultimaker 3 profiles.

The parameters that are "leaves" have no settings directly below them that are indented further. It's easiest to see by making all settings visible first. They are also the settings in fdmprinter.def.json that don't have any "children" property.

BenRoe, 2018-08-27:
Hello,
i try to change the layer_height via CLI.
```
./CuraEngine slice -v -j /path/to/custom_printer_settings.def.json -o /path/to/output.gcode -l /path/of/file.stl -s layer_height=0.3
```
But it has no effect. Any tips?

BagelOrb, 2018-08-28:
You are passing that setting to the last loaded object, while you should be passing it to the global settings container. Put the `-s layer_height=.` before the `-l object.stl`.

vatansbeytr, 2019-07-22:
`CuraEngine slice -v -p -j "C:\Program Files\Ultimaker Cura 4.1\resources\definitions\fdmprinter.def.json" -o "C:\Users\xD_xD\Desktop\deneme.gcode" -s infill_sparse_density=50 -l "C:\1.stl"`  same here I just try to change to value of density but It doesn't even work. Hey any have a valid def.json file already calibrated because every time I used the console even by default setting giving me different time from cura gui

BagelOrb, 2019-07-22:
@vatansbeytr `infill_sparse_density` is a parent setting. You can only set the child settings via the command line. You will have to set `infill_sparse_line_distance` or whatever the child setting is called.

vatansbeytr, 2019-07-22:
@BagelOrb I tried to change the settings from def.json file inside fdmpeinter but it doesn’t effect the result.

Ghostkeeper, 2019-07-22:
Even changing the infill_sparse_density in fdmprinter.def.json won't have any effect if you're slicing via command line. CuraEngine never reads that setting. Cura's front-end does, and then puts the correct setting value to reach the desired density in the infill_sparse_line_distance setting.

vatansbeytr, 2019-07-22:
@Ghostkeeper any suggest for changing infill because I don’t have any clue about it.

Ghostkeeper, 2019-07-23:
Try this:

    CuraEngine slice -v -p -j "C:\Program Files\Ultimaker Cura 4.1\resources\definitions\fdmprinter.def.json" -o "C:\Users\xD_xD\Desktop\deneme.gcode" -s infill_line_distance=0.8 -l "C:\1.stl"