Ultimaker/CuraEngine #537: If i add '\n' to my start or end gcode nothing happens

brunoaandrade, 2017-06-21:
I'm using the following command line:
```
"C:\Program Files\Cura 2.6\CuraEngine.exe" slice -j "c:\Program Files\Cura 2.6\resources\definitions\prusa_i3_mk2.def.json" machine_end_gcode="M104 S0 ; turn off extruder\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\nG1 X0 Y210; home X axis and push Y forward\nM84 ; disable motors" -l Marvin.stl -o Marvin.gcode
```
My gcode file contains this end gcode:
```
M104 S0 ; turn off extruder\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\nG1 X0 Y210; home X axis and push Y forward\nM84 ; disable motors
```
instead of:
```
M104 S0 ; turn off extruder
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X0 Y210; home X axis and push Y forward
M84 ; disable motors
```
My end gcode is just a string, the line breaks ('\n') are being ignored. If i want to make a change my only option is to change the prusa_i3_mk2.def.json file directly?

BagelOrb, 2017-06-22:
Just make this your command line call:
```
"C:\Program Files\Cura 2.6\CuraEngine.exe" slice -j "c:\Program Files\Cura 2.6\resources\definitions\prusa_i3_mk2.def.json" machine_end_gcode="M104 S0 ; turn off extruder
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X0 Y210; home X axis and push Y forward
M84 ; disable motors" -l Marvin.stl -o Marvin.gcode
```

closed 2017-06-22 by BagelOrb

dombmann, 2020-02-05:
Instead of opening a new Issue I am asking here.
Why is the -s overwrite Parametere not needed here ? 
`machine_end_gcode="M104....` instead of `-s machine_end_gcode="M104....`
My main question is  the following: 
How should the machine end/start code be formatted when CuraEngine is called from a (windows) bash script ? I want to slice .stl using the commandline style but (since the huge ammount of -s paramters) CuraEngine is called from within a bash file.

BagelOrb, 2020-02-05:
Yeah I forgot to type the '-s'.

I don't know how to do that from bash. 
Maybe this can help
https://unix.stackexchange.com/questions/481794/how-shall-i-pass-a-newline-character-to-a-command-in-a-script