Ultimaker/CuraEngine #544: Support - Use Tower bugged?

megablue, 2017-06-30:
Cura 2.6.1
When "Use Tower" is enabled. Cura will try to generate tower outside and underneath a hole where nothing requires supporting... 

[image]

BagelOrb, 2017-06-30:
It is actually trying to support the small bit of support inside the hole a
bit higher up, but the support tower gets too small and gets pushed away
from the model at the top of the tower, which makes it unprintable thin.

The tower is generated for support, but the tower isn't fully built.

This is a limitation of how it is currently implemented.

We'll have to see when we can fix something like this and how big of a
problem this is.

alekseisasin, 2018-01-25:
Hello @megablue    Sorry for waiting long time, for now this issue can be solved if you will set setting
¨Support Horizontal Expansion¨ value to 0.