Ultimaker/CuraEngine #547: [master] Broken i386 builds

thopiekar, 2017-06-30:
I'm curious what it is, but I don't have the time at the moment.
One thing I can say and what you can see easily on your own is that other architectures, like armhf, amd64 and powerpc work..

https://code.launchpad.net/~thopiekar/+archive/ubuntu/cura-master/+build/12832047
https://code.launchpad.net/~thopiekar/+recipe/cura-engine-master-daily
https://code.launchpad.net/~thopiekar/+archive/ubuntu/cura-master

BagelOrb, 2017-06-30:
I don't really know how to read those logs. Is this about the segfault in the StringTest?

thopiekar, 2017-07-03:
Yes, what you can see is the full log of the build. At the beginning there is only a very minimal installation and then it installs all needed packages. This way you can even track what has been installed in which version.
Yes, the only thing I see is the SegFault, too. I know to less about C programming to tell you how it comes that other architectures work and i386 not.

BagelOrb, 2017-07-03:
I discussed with @Ghostkeeper : We don't support 32bit anymore, so we don't really have the means to fix this.

closed 2017-07-23 by BagelOrb