Ultimaker/CuraEngine #554: Feature request: Attempt to print small features

AlucardF0X, 2017-07-06:
Many times I make a model, it has small features that are close to the size of my nozzle.

Nozzle is 0.4mm
Line width is 0.4mm
Layer height is 0.2mm

The feature size is 0.6mm and cura refuses to slice it. 
It just leaves a big blank there
'Fer christs' sake, it's a ADDITIVE process, not a "im too lazy to make anything", I can just file it down to appropriate dimension, I just need the printer to actually print it first.

I can attach screenshots and a STL of what I mean, if you need additional information. 

This happens because Cura tries to slice the model, tracing the model WALLS to make SHELLS.
And thin parts of the models only have room for ONE shell (not on each side, both sides, one shell total, period). 
I have a plethora of 3D models that can't be printed in cura because of this, and I need to resort to Slic3r's "detect thin walls"... BUT SLIC3R IS SUCH A PAIN TO USE AAARGH!

I'd love cura forever if it gets 'detect thin walls'.
It can clearly detect 'em and skip' em. 
And cura can clearly do thin walls, like with zigzag supports. 
Even better if it underextrudes super thin features, to attempt printing super thin! Go cura, I know you can do it! It's better than printing NOTHING...

And I'd donate a hundred bucks to the cause if it also gets adaptive layers, like Prusa research added to their program (although I understand this feature is particularly hard to implement as it conflicts with, literally every other part of the code).

BagelOrb, 2017-07-12:
Currently working on a PR for this feature..

AlucardF0X, 2017-07-12:
I think the easiest way to implement this is to allow the shells to intersect eachother, and then the intersections are detected in a 2nd step to avoid overextrusion, they are either lowered in extrusion (to make very tight seal where the two walls merge into one), or alternatively the 2nd wall is completely removed.

I think this is how Slic3r does it. And I don't think there's a better way, unless you modify the entire slicing code to check opposite walls at every step before generating a wall. That'd be slow.

BagelOrb, 2017-07-12:
Cura is already doing that. The problem is with walls equal to or smaller
than the nozzle size.

For that last problem I made a hack to make lines which generally follow
the wall, but not always because that's really hard to implement.