Ultimaker/CuraEngine #558: Feature request: Layer preview override thickness - tubes too thicc

AlucardF0X, 2017-07-12:
My 3D Printer is running a 5:1 geared extruder.
To compensate for that I need to do 355% flow (remember, drive gears are larger than a dot).

But this makes Cura output some gnarly thick lines that make it hard to discern what's being printed. 

In Cura 1 / classic layer preview - this "bug" made the top layer a giant and I mean GIANT rectangle, absolutely no layer visible. 

**<Speculation>**
I tried to see if there's an easy fix for this, but it seems Cura outputs the gcode paths directly with the thickness combined in 'z' shapes like the classic layer preview, instead of outputting clean lines that accurately represent the gcode path. These Z's are then turned to rectangles in classic, and tubes in the new layer preview. (Tubes look awesome)
This means the flow is never directly referenced anywhere. 
**</Speculation>**

There's a hack I can do to temporarily fix this for time being which I'll document as follows:
At file "\Program Files\Cura 2.6.1\plugins\LayerView\layers3d.shader"
At line 133 (Notepad++)
There is this line:
             size_x = v_line_dim[0].x / 2 + 0.01;  // radius, and make it nicely overlapping
I can fix my 'tubes too thicc' problem by modifying the line to this:
            size_x = v_line_dim[0].x / 7.1 + 0.01;  // radius, and make it nicely overlapping

What I've done is changed the output tube thickness. 
By default it takes the rectangle width, divides it by 2 to get a tube radius.
I've additionally added a "divide by 3.55" which is my extrusion multiplier, resulting in div by 7.1
This fixes my problem and sets "355 flow" as the normal "100".

I think a simple fix as so: In layer preview view (menu), add a 'default flow: 100%', which will be then uploaded to shader code as preference, and it will multiply the "2" in the code line I've shown. 
In my case I'd simply need to make it 355%, and it'll miltiply the 2 by 3.55 and fix my issue not-hackily. 

I've added pictures demonstrating the problem. The one that looks wrong is with the '2'. The one that looks fine is with the '7.1' (surround audio joke)
[image]
[image]

.

nallath, 2017-07-13:
Shouldn't your firmware be changed instead? We have a similar issue with the UM2 vs the UM2+, which is fixed by changed the steps per E on the firmware side.

AlucardF0X, 2017-07-13:
My firmware is a stupid china version that doesn't remember it's step per E between resets... Do you know how much of a pain it would be to keep setting the steps every printer run? 

The modification I suggested to the software is extremely easy to add. I already implemented it half way. 

Some filaments require underextrusion as they come out liquidy, but the software (cura) is going to show them as if they're being extremely underextruded. This is a RENDERING error, not a firmware error.

BagelOrb, 2017-07-14:
You can also change the filament diameter. That can increase the E steps without altering the actual flow you print with.

If you increase the flow the lines will become thicker, so they must be visualized thicker. We're not going to change that for your "stupid china firmware version".

AlucardF0X, 2017-07-14:
... really? 
that's pretty hostile
are you even going to provide the calculation one would use to alter the filament diameter?
no?

Just closing this feature request even though it's a simple 2 minute UI implementation and shader edit? 

Are you making cura proprietary or something?

BagelOrb, 2017-07-14:
I'm sorry if I seemed hostile. I had a run through a lot of issues today, so my answer was not that well thought through.

This issue contains two parts:
1. You have the problem that your firmware uses the wrong E steps per mm filament
2. You suggest to change the visualization of lines such that they are unaffected by the flow.

My response:
1. You should fix the firmware, but as a hack you can increase the filament diameter to sqrt(3.55) it's actual diameter.
2. We will not do this because adjusting the flow actually makes lines thicker, so they should appear as such. If you compensate your 3 steps per mm by the suggestion in 1, you won't have the problem any more.

I hope this helps

AlucardF0X, 2017-07-15:
did you... read my suggestion at all? 

I said the shader has an easy spot for including the behavior I was looking for.
This inclusion would incur a very low performance overhead - just one extra multiplication, which can be optimized away if it's pre-multiplied externally.

This value would be stored as a numeric box in the layer view settings panel. A preference.

I already implemented 90% of it, all that's left is just updating a shader variable from a form element, which is a 5 minute fix. I just don't want to hack this in EVERY UPDATE OF CURA EVER?

This isn't exactly a problem with e steps per mm, or firmware. I included a PLANETARY REDUCTION. Firmware isn't exactly using wrong e steps per mm filament. 

Thanks for the actual filament fix suggestion. Although it's incomplete?
HackyDiameter = ? sqrt( flow ) ?  

And for the visualization, people might want to, oh i dont know, SEE THE PATH CLEARLY? Even if they're over-extruding? It hardly includes clarity of any kind other than polluting the view with extra thicc lines. If cura included variable flowrates, I would concede but it doesn't at present... 

I have a very popular china clone, this isn't a one-off scenario.

I seriously don't understand the hostility to this idea which takes 2 minutes to implement. Did I hit a nerve? Are you jealous of how genius this simple fix is? Are you proud of the fat pipes that make it impossible to see what's going on? 

I really wish to get a 2nd opinion on this by someone else, see if they also think this feature should be included or not. 

God you guys are making me want to make a proper procedural slicer that doesn't suck weird misshapen balls...

BagelOrb, 2017-07-15:
> God you guys are making me want to make a proper procedural slicer that
doesn't suck weird misshapen balls...

And you ask whether you hit a nerve..
Well, I got agitated by the fact that instead of asking our help with your
problem, you came up with your own solution (adjusting flow) and then
blamed us for the limitations of your solution.

I work at Ultimaker, so I have no incentive to do work to support your
Chinese clone.

As for the second opinion: if you implement the remaining 10% and make a
pull request then someone else will be looking at it.

Ghostkeeper, 2017-07-17:
I have no qualms against implementing a preference for adjusting the line widths visually, like Repetier-Host has. It doesn't need to be an actual setting, just a visibility thing.

(In other news, we really do need to make a preferences page specifically for layer view by now...)

Ghostkeeper, 2017-07-17:
That would actually also be sort of useful when printing high infill densities with combined infill layers. It suffers from the same problem.

AlucardF0X, 2017-07-17:
Oh, look, there's people who agree with my idea, and actually provided actual help for a workaround. Thanks Ghostkeeper. 

Actually I already "hacked" the shader file, as detailed in my first post, so I don't need to do the filament diameter thing. I use 3mm filament diameter. I just wanted a diameter tweaking setting for the preview. If 

I have a green go-ahead I could implement it myself, but only if there's 100% guarantee it'll actually be included. I hate github exactly because all the good work gets rejected by cunts like BagelOrb, but all the bad useless things get put in and make the software bloated, like happened with Firefox multiple times. 

Fck you BagelOrb and your hostile denial of my idea without any valid criticism and closing my thread before anyone else even replied to it. Personal grudge or something?

nallath, 2017-07-17:
Excuse me? You want something from people and if you don't imediately get what you want your response is to call them cunts? How exactly do you think that this improves the odds of the thing that you want ever getting done? 

I might have considered helping you before, but if you feel that this is a normal way of talking to people, i pretty much don't feel like this.

I think the cricism was valid. I wholly agree with the points made. The steps per mm *are* wrong. If you tell it to extrude 1 mm, does it happen? No, eg; the steps are wrong.  He wasn't the only one that replied. I did too.

Ghostkeeper, 2017-07-17:
Tip from @daid he told us here in the office: Add [M92](http://reprap.org/wiki/G-code#M92:_Set_axis_steps_per_unit) to your machine start g-code to adjust the steps per mm. That's acutally a very elegant solution.

closed 2017-07-17 by nallath

AlucardF0X, 2017-07-17:
That's another great suggestion, Ghost. I might've actually used the M92 as start code if I didn't already modify the shader code.

But there's a small problem with it - it's machine specific, you'd need to check the machine current steps per mm, calculate the new steps per mm... While in my situation, I just multiply the flow directly and it works on any and all machines with the 3:1 reductor gear, regardless of what stepper motor they have.

AlucardF0X, 2017-07-17:
On a side note: 
1. This is a feature suggestion. 
2. I didn't behave professionally.
3. So... the cunts decide to abuse their admin power to close the FEATURE suggestion, because of the SUGGESTER. 

To summarise: 
The cunts closed the IDEA not because the idea is bad, but because they dont like the PERSON. 
So you'd get in the way of progress, just because emotional reasons? That's sad. Very sad. SJW-level sad. Regressing because it hurts your feelings. :(
That's what I'd call bonafide admin abuse.

BagelOrb, 2017-07-17:
You've said in all caps that it's a "VISUALIZATION ERROR".

Moreover I've given reasons why I think such a feature is a bad idea.

Besides, closing a feature request doesn't necessarily mean it's a bad
idea. It just means that there are better ideas to work on the next couple
of years. There's loads of ideas around.

daid, 2017-07-18:
Hello. Nr1 CUNT here. AKA, the Cura guy, who has been called more names during the development of Cura because of choices then you can throw at me. So your verbal abuse of my colleagues doesn't really surprise me or surprise me.

First off. We are allowed to do whatever we want with our software. It is our software. Don't like it? Fork it, suggest things, send pull requests. But don't verbally abuse us, as in the end, that makes you a lesser person in life in general.

And we are not supporting your abuse of the flow settings to fix the shortcoming of your machine configuration. The M92 Exxx is the generally accepted solution for machine variations on extrusion for the past 5 years. Has nothing to do with the person who suggest is acting like an total ass. But that doesn't help. It has to do with what is responsible for what. GCode asks for 1mm movement, machine should give 1mm movement. Your machine does not, fix for that is adding some start code that it does. Not a hack in the visualization of GCode, and abusing a setting for what's it's not intended for.

AlucardF0X, 2017-07-18:
Hello Mr. Nr1CUNT 😃  . I was calling BagelOrb and nallath cunts for the simple reasons they didn't even complete their "alternative solution" and closed my feature request without any criticism I would consider valid. Surely they aren't going to commit suicide over it? 

Thanks for your suggestion earlier as Ghostkeeper kindly forwarded. I however added what I consider to be valid criticism, as M92 is a printer-specific fix. If I have multiple printers with different steppers, I would need to adjust this setting for each and every printer with a non-trivial formula. Setting it per-machine as cura allows would be a nice fix since my machine doesn't have setting memory. 

I know you are allowed to do whatever you want, I never stated what you have a legal right to do and not do; I simply expressed my disdain for your two colleagues who kicked my idea out of the world before anyone other than them could even check the idea. Negative criticism is often rash and illogical. Ghostkeeper expressed an interest in this feature, for example. If even ghost stated the idea is useless, I would've conceded. 

I wouldn't call what I did verbal abuse, I was venting my frustration that your colleagues in a hostile way closed my request without considering it at all, as if this is the 50th time they close a similar request? 
I'm very well aware verbal abuse makes one a lesser person, but that doesn't matter in the business world, this isn't a social gathering where you gather the "elite" and look down on the "lesser" ... this is a site for suggestions and software improvement. 

I understand you consider the flow setting to be abuse. As ideally the setting was designed to alter the software-produced flow and correct it for slight under/over extrusion. In my case it isn't overextrusion, it's compensation for another setting altogether, which I admit, should be done elsewhere, either in steps/mm (which in my case will produce erroneous extra step every 3.5 turns due to the imperfect 355% flow) or as a secondary step, or even as an alteration of the filament milimeter (which will screw up filament statistics). 

The reason I suggested it is done within the flow setting, is simply, because it's the nicest setting that already exists, so my needs can be met with little to no modification of the software. 

What I requested isn't "support for abusing the flow", I requested a feature for adjusting the preview's thickness of lines. I already hacked this manually as detailed in the original post. I was requesting that simple settings for the layer preview are added, so they could support cases where the extra flow would make it hard to understand what's going on in the layer preview. 

In Cura 1, a flow of just 110% made the layer preview go insanely thick, that's why I switched to Simplify3D until Cura2 came out.

AlucardF0X, 2017-07-18:
Also, I abuse a lot of settings of the slicer to produce amazing results.
I can guarantee you that atleast 20% of the settings on my end are being abused heavily to produce alternative results. Using software the way the developer intended is often "bigoted" and can lose customers. that's why slic3r is still a popular option, because it doesn't force you to use the settings as the developer sees fit. It's merely a software that lets you use its features, to your creative prowess. 

Instead of having an angry fit on me here, you could've spent the time to implement the feature I requested. For example. Am I wrong? 

I said I could easily make a fork/pull and implement it, if it can be guaranteed that you won't just reject the pull request and waste my time having developed the fix. I've had that moment on git enough times to realize I need an inside-guy to greenlight my changes before I even develop them.

Ghostkeeper, 2017-07-19:
If you go to "Manage Printers" (under the printer selection drop-down), select your printer and then click on "Machine Settings", you'll get a menu that allows you to change the start g-code among other things. This is a good place to add the `M92` code.

The start g-code is a setting specific to the machine, but so is flow. If you swap the machine in Cura, you have to set the flow again.

The way I see it, a professional should never use the Flow setting but Flow is still useful as a quick fix or for someone who doesn't know all of the intricacies. You should be able to modify it, but obviously the visualisation will be updated accordingly. I think that's correct.

A generic modifier for line widths in the visualisation could be useful for all sorts of reasons so I made a ticket for that in our internal tracker (even though the Github issue is closed now). It's up to our product manager to decide if and when we're going to implement that. There's a chance that the ticket gets closed after 3 months because there was not enough desire for it and there's bigger fish to fry. We'll see.