Ultimaker/CuraEngine #559: GCode always ";TIME:6666 ;Filament used: 0m"

StuCoder, 2017-07-15:
I'm using CuraEngine with command line, generate  a gcode file. 
`CuraEngine.exe slice -v -j resources\kossel_mini.def.json -o test.gcode -l test.STL`
The beginning is always like this

```
;FLAVOR:RepRap
;TIME:6666
;Filament used: 0m
;Layer height: 0.2

;Generated with Cura_SteamEngine 2.6.2
```

but command line output:

```
Gcode header after slicing:
;FLAVOR:RepRap
;TIME:32577
End of gcode header.
Print time: 32577
Print time (readable): 9h 2m 57s
Filament: 60174
```

I don't know why.

BagelOrb, 2017-07-15:
Because the header is written before the engine knows what it's going to
print.
The header that is shown via the command line is what it should eventually
be.

StuCoder, 2017-07-15:
I want the gcode file to contain the correct header.
What should I do to get the correct gcode file through the command line?

BagelOrb, 2017-07-17:
I don't know. Perhaps you can write some script which parses the gcode and
the output and then places the correct header at the start.

Ghostkeeper, 2018-02-21:
Our project manager removed this from our planning because it was not deemed important enough.

devangsaini123, 2018-07-16:
When I capture the command-line output, it is still not able to collect the final header. I just want to capture the final filament_used and print_time, which is printed post the line "Gcode header after slicing:". Calling it in the following fashion,
`output = `CuraEngine slice -v -p -j printer.def.json -l model.stl`.split("\n")`
Am I missing something here?

Beherith, 2018-07-30:
The only way I was able to capture it, is by redirecting stdout to a file, and stderr to a file. The results were in the stderr file. E.g. 
CuraEngine slice -p -o AlcoveWall_v2.gcode -j fdmprinter.def.json  -s mesh_rotation_matrix="[[1,0,0],[0,1,0],[0,0,1]]" -v -l quick_retract_test.stl 1> stdout.txt 2> stderr.txt

Ghostkeeper, 2018-09-17:
The "g-code header after slicing" is indeed more intended as a debugging output rather than to be put in the g-code. The firmware doesn't read the header there anyway.

emrem6, 2020-06-17:
> The only way I was able to capture it, is by redirecting stdout to a file, and stderr to a file. The results were in the stderr file.
> ALSO: dont forget the -v tag for verbosity, otherwise it gets silenced. Kind of like this software wasnt meant to be used this way...
> E.g.
> CuraEngine slice -p -o AlcoveWall_v2.gcode -j fdmprinter.def.json -s mesh_rotation_matrix="[[1,0,0],[0,1,0],[0,0,1]]" -v -l quick_retract_test.stl 1> stdout.txt 2> stderr.txt

How did you managed that to write the stderr or atdout to a file? This is what I need to do currently?

Ghostkeeper, 2020-06-18:
The `-o AlcoveWall_v2.gcode` option basically writes the stdout to a file. Or you can use the `1> stdout.txt 2> stderr.txt` like Beherith did.