Ultimaker/CuraEngine #560: Bug - No-skin combing always retracts within infill mesh

smartavionics, 2017-07-16:
I have a print which uses infill meshes to beef up certain areas and I've noticed that when combing is set to no-skin, the travels within the infill areas (that have skin and wall thicknesses set to 0) are all retracts. See this pic:

[image]

I've spent a while looking at the code but can't figure out what's wrong. As an experiment, I  stopped it adding the infill mesh's infill_area to the combing boundary in LayerPlan::computeCombBoundaryInside() and then the travels don't use retract but that's not a good solution because if the infill mesh has any skin (why would anyone give an infill mesh skin/walls anyway?) the travels also don't retract when printing the skin parts which is wrong when combing mode is no-skin. Perhaps someone who is familiar with this code and take a look as I'm wasting too much time on this!

smartavionics, 2017-07-16:
When combing is enabled for all surfaces, it looks like this:
[image]

smartavionics, 2017-07-16:
Tim, the following appears to work but I'm not sure it's the right fix! What it does is when computing the comb_boundary, if the mesh is an infill mesh, it adds the skin part polygons to the comb boundary and not the part's infill_area. My reasoning is that adding the infill_area from the infill mesh makes a hole in the infill_area of the non-infill mesh but you only need to make holes for the skin parts (if any!) in the infill mesh.

Whaddyathink?

```
Polygons LayerPlan::computeCombBoundaryInside(CombingMode combing_mode)
{
    if (combing_mode == CombingMode::OFF)
    {
        return Polygons();
    }
    if (layer_nr < 0)
    { // when a raft is present
        if (combing_mode == CombingMode::NO_SKIN)
        {
            return Polygons();
        }
        else
        {
            return storage.raftOutline.offset(MM2INT(0.1));
        }
    }
    else 
    {
        Polygons comb_boundary;
        for (const SliceMeshStorage& mesh : storage.meshes)
        {
            const SliceLayer& layer = mesh.layers[layer_nr];
            if (mesh.getSettingAsCombingMode("retraction_combing") == CombingMode::NO_SKIN)
            {
                if (mesh.getSettingBoolean("infill_mesh"))
                {
                    for (const SliceLayerPart& part : layer.parts)
                    {
                        for (const SkinPart& skin_part : part.skin_parts)
                        {
                            comb_boundary.add(skin_part.outline[0]);
                        }
                    }
                }
                else
                {
                    for (const SliceLayerPart& part : layer.parts)
                    {
                        comb_boundary.add(part.infill_area);
                    }
                }
            }
            else
            {
                if (mesh.getSettingBoolean("infill_mesh"))
                {
                    continue;
                }
                layer.getSecondOrInnermostWalls(comb_boundary);
            }
        }
        return comb_boundary;
    }
}
```

BagelOrb, 2017-07-17:
The infill mesh should fully be classified as infill. Its skin can be
combed over because it is never visible.

Perhaps the comb boundary uses the infill_area_own rather than the full
infill_area. Or maybe the Comb::getSecondWalls adds the walls of the infill
mesh, which polygons are then interpreted as holes in the comb area of the
outer mesh.

smartavionics, 2017-07-17:
> The infill mesh should fully be classified as infill. Its skin can be combed over because it is never visible. 

Well, if the non-infill mesh had no skins but the infill mesh did, they would be visible!

Anyway, I have a PR that fixes this problem which is based on what I have above, I'll submit it for your contemplation.

BagelOrb, 2017-07-17:
You didn't address your own issue. In your PR the skin area of the infill mesh isn't taken into account.
I still think that that's really an edge case which doesn't matter that much, though.

smartavionics, 2017-07-17:
Sorry, I don't understand your last comment. Why isn't it taken into account?

BagelOrb, 2017-07-17:
You're not looking at the difference between infill and skin in case of infill meshes. Your change makes the algorithm only look at the outlines of infill meshes. Those outlines enclose both the infill and the skin (and the walls) of the infill mesh.

smartavionics, 2017-07-17:
> You're not looking at the difference between infill and skin in case of infill meshes. Your change makes the algorithm only look at the outlines of infill meshes. Those outlines enclose both the infill and the skin (and the walls) of the infill mesh.
Hugh? I don't think we are on the same wavelength, here.

BagelOrb, 2017-07-17:
Let me respond in the place in the PR.

smartavionics, 2017-07-17:
Closing, discussion moved to #561

PR #561 (2017-07-19): Correct the infill mesh areas added to the comb boundary when combing mode is NO_SKIN.
It was adding the area of the infill mesh's infill but this just makes a hole in the
comb boundary the shape of the infill mesh. Now, it adds the outlines of the skin areas
in the infill mesh (rather than the infill area).

Fixes #560.
closed 2017-07-17 by smartavionics