Ultimaker/CuraEngine #563: Cura 2.61, Surfave mode and Fuzzy don't work on hole

yanivg101, 2017-07-20:
I have this two models:
Heart and Cylinder
I tried to 3D print the Cylinder on Fuzzy mode, only one perimeter and the Heart on Surface mode.
https://drive.google.com/drive/folders/0B6cDIjXDeHf4X2xDUVdFekNjR0E?usp=sharing

this is the settings i'm using and this is the results (I wanted them to be merge but for this example I separated them so you can see the the Cylinder model looks wierd)
[image]

There are two version of each file, one of them (the short names) have small wall on the edge and the others (Long names) are without this walls.

When I checked "Keep Disconnected faces" I got this results but it's fill the hole:
[image]

and when I combine them I got this results with additional line on the perimeter:
[image]

BagelOrb, 2017-07-20:
I'm sorry for this bug, we know that Surface mode is quite buggy at the moment.

Please don't combine Keep Disconnected Faces with Surface Mode Both. They both control the same thing. Maybe try disabling Keep Disconnected Faces but enabling Surface Mode Both.

I don't think we're going to solve this bug any time soon, because almost nobody would ever combine these two excentric features and there's other bugs to fix.

BagelOrb, 2017-07-20:
The X-ray view should be red around the hole, because surface mode is meant for meshes which don't describe a solid. Is that the case?

yanivg101, 2017-07-23:
Yes, It's red:
[image]
I just want to know if there any way to produce such model.

yanivg101, 2017-07-23:
I will try to make only the Cylinder Solid and set it to Normal and Fuzzy and the Heart to Surface:
Now it's look like this:
[image]
[image]
[image]

I hope the model will 3D Printed well

BagelOrb, 2017-07-23:
What you are trying is currently just not really supported.

There's a feature branch on its way to being merged which makes it possible
to provide a black and white texture which determines the fuzziness at each
location, but there are a lot of frontend features required to make such a
thing work. It'll be at least a couple of months before that feature is
incorporated into Cura.

yanivg101, 2017-07-23:
Can you use OBJ with Texture for that? it will be awesome feature!!!

BagelOrb, 2017-07-23:
Yes, hopefully that will be possiblein Cura 2.8 or 2.9, but I cannot
promise anything.

yanivg101, 2018-01-29:
Hi,
There is any progress in this feature?

BagelOrb, 2018-01-29:
Nope :/