Ultimaker/CuraEngine #570: Raft Fan Speed ignored

bekriebel, 2017-07-27:
With Cura 2.6.2, Raft Fan Speed is being ignored. Setting Raft Fan Speed or any of the sub-settings still results in GCode with M107 (fan off) being issued before each raft layer.

Repro steps:
1. Set Build Plate Adhesion to Raft
2. Set Raft Fan Speed to 100
3. Save GCode
4. Examine GCode and see M107 being called before each raft layer

Also reported on the forums: https://ultimaker.com/en/community/50306-cura-raft-fan-speed-settings-bugged

BagelOrb, 2017-07-28:
Will look into it today before my holidays kick in.

BagelOrb, 2017-07-28:
Confirmed.
Fixing it.

BagelOrb, 2017-07-28:
Fixed

closed 2017-07-28 by BagelOrb

BagelOrb, 2017-07-29:
No thank you for reporting this here.
Otherwise it wouldn't have been fixed in Cura 2.7!