Ultimaker/CuraEngine #577: Too much overlap compensation?

smartavionics, 2017-08-02:
Take a look at this example, the round feature consists of two outside walls. The outermost one was printed first and the innermost printed later and it has been overlap compensated down to almost nothing where it touches the outermost wall. In fact, it is drawn with such a thin line that you can hardly see it in this image.

[image]

But the (inside) outline that it should have is actually the same as the outline on the layer below which is quite visible. I realise that the centre line of the very thin wall may not be lined up with the centre line of the corresponding wall below so it's hard to tell exactly but it looks to me that the thin wall is much too thin!

I don't think it's just a glitch in the layer view because this is what S3D shows the gcode to be:

[image]

This is just one example, I can see many other places where it looks like the

smartavionics, 2017-08-02:
Well, of course, I couldn't just leave this problem unexplored so I have wasted a few hours trying to understand the marvels of the overlap compensation code. After making some changes in WallOverlapComputation::getApproxOverlapArea() it's now producing more sensible results:

[image]

It's hard to see in the image but the wall in the centre is actually wider than it was before. I also have other examples which have improved.

Old code:
[image]

New code:

[image]

smartavionics, 2017-08-03:
I'm continuing to work on this but don't really understand how a lot of the code actually works.
Currently, I am getting best results if I disable the processing of zero-length segments (aka points) in handlePotentialOverlap(). Could someone who is familiar with this code please explain what the zero length segments represent and what effect they think those zero length segments should have in the overlap area calculation.

Ghostkeeper, 2017-08-17:
I've seen a similar effect sometimes even with simple cubes but it's fairly hard to reproduce.

smartavionics, 2017-08-17:
I am currently using a heavily modified version of wallOverlap.cpp and I believe it's performing better than the original but I would be the first to admit that I don't really understand how it works! I could create a PR so you can see what I have done but it's very much WIP.

BagelOrb, 2017-08-17:
Zero length segments are locations with 100% overlap.

They help in overlap compensation on sharp corners where one polygon
segment ac overlaps with the next one ce.
The locations b and d on AC and CE are identified where the overlap is just
0%. Then they are linked with a segment of say 0.4 mm.
Another link is created between c of AC and c of CE. The length of this
link is 0.

Then the normal overlap area calculation can be used with the .4mm link and
the 0mm link.

Ghostkeeper, 2018-02-12:
Our project manager removed this bug from our planning because it wasn't deemed important enough.

smartavionics, 2018-02-12:
Yeah, that doesn't supprise me. I know you guys don't give a damn about the rotten overlap compensation.

BagelOrb, 2018-02-12:
It's more that the code is too complicated to fix easily. I should have
added way more documentation!

It might even be the case that the algorithm is fundamentally flawed.

smartavionics, 2018-02-12:
As you may remember, I have made some attempt at improving (IMHO) the overlap compensation (it's #586) but have ground to a halt because I simply don't understand how a lot of the existing code works (all that stuff about links, etc.) I am still using my modified version and perhaps, one day, it can be progressed.

closed 2018-04-15 by smartavionics

AbeFM, 2018-06-19:
Did your modifications get rolled in? This behavior has had me running to Slic3R several times.

Tinchus2009, 2018-11-17:
Hi. I have been following this issue. The overlap compensation is working ok, still is considered to be experimental somehow or it has been removed?