Ultimaker/CuraEngine #578: PathOrderOptimizer::getClosestPointInPolygon() failures

smartavionics, 2017-08-04:
With the existing code, I get this when I specify a z-seam hint:
[image]
If I comment out the is_on_inside_corner_score stuff (i.e. don't try to detect the inside corner), I get this:
[image]
This begs the questions, how useful is the inside corner detection and is there a good argument for not doing it?
I'm going to disable it for now and see if I can detect any situations where it would have been useful.

sco01, 2017-08-16:
Does this change fix [397](https://github.com/Ultimaker/CuraEngine/issues/397) as well?

Ghostkeeper, 2017-08-23:
This is not a change, it's just a bug report. I don't believe this is the same bug as #397 since that bug is about the inner walls and this is about the outer walls.

I think there is a mistake in the inside corner detection.

I do think inside corner detection is very useful! I try to get my seam on the inside corners whenever I need to make something that looks pretty.

smartavionics, 2017-08-23:
Morning. The whole point of specifying the z-seam is so that you get a nice regular seam. The inside corner detection needs to be made more robust so as not to introduce irregularities as shown above.

BagelOrb, 2017-08-23:
I agree. Good catch!
Perhaps we should simply snap `is_on_inside_corner_score` to zero if it is already close.
If the angle is within 45 degrees off of a straight line the score should snap to zero.
Shouldn't be too difficult to fix.

smartavionics, 2017-08-23:
Shall, I have a go at implementing that, seeing that I'm the one who's complaining!

BagelOrb, 2017-08-23:
Would be nice. A pull request will make it faster into the build than when we file a bug ourselves ;)

snapping to zero is not really the solution, though. Let me think for a minute.

smartavionics, 2017-08-23:
OK, when you think you know what's needed, let me know and I will implement and test.

BagelOrb, 2017-08-23:
Maybe something like:
```
float right_angle = 2 - LinearAlg2D::getAngleLeft(p0, p1, p2) / M_PI;
float is_on_inside_corner_score = right_angle * right_angle * 5000 * 5000; // prefer inside corners
```

smartavionics, 2017-08-23:
So this is using Tim's suggested code above. The z-seam origin hint is 0, -50 so that's off the bottom of the screen. As you can see, it didn't work for this example (I was hoping the seam would be in the bottom of the little indent):
[image]

smartavionics, 2017-08-23:
I had a thought! Does this code need to take into account whether the prev_point is inside or outside of the polygon? If it's inside the polygon you would want to decrease the distance for the inner corner and if it's outside, you would need to increase the distance for the inner corner?

BagelOrb, 2017-08-24:
No it doesn't need to. The direction of hole polygons is the other way around as outline polygons. The left angle is always the angle taken across the inside of the polygon. Outside polygons are counter clockwise, while hole polygons are clockwise - if I recall correctly.

BagelOrb, 2017-08-24:
You might also put something in there like:
```
if (right_angle > 0.5)
{
    right_angle = 1.0;
}
```

Maybe you're right and the is_on_inside_corner_score should by multiplied by -1. I might be confusing CCW and CW here...

smartavionics, 2017-08-24:
My current thinking is that if the angle is less than 180 deg it's a convex corner so leave the distance unchanged. When the angle is > 180, it's a concave corner and we want to boost the point's chances of being chosen. The more concave the corner, the more we like it. So current WIP is this:
````
       int64_t dist = vSize2(p1 - prev_point);
        double corner_angle = LinearAlg2D::getAngleLeft(p0, p1, p2) / M_PI; // 0 -> 2
        if (corner_angle > 1)
        {
            dist -= (corner_angle - 1) * dist / 10; // prefer inside corners
        }
        if (dist < best_point_score)
        {
            best_point_idx = point_idx;
            best_point_score = dist;
        }
````
My weird mushroom example with a rectangular slot in the side comes out well as you can see in the following picture. Will do some more testing and see where it breaks...
[image]
The z-seam hint is at (5, 100) so it's well outside of the object.

smartavionics, 2017-08-24:
I think a problem with the previous implementation is that the fixed size adjustment based on 5mm doesn't take into account the distance from  prev_point which can be anything so my code above uses 10% of the distance from the prev_point to p1.

smartavionics, 2017-08-25:
I've played with this some more and it works pretty well. The / 10 is better as / 50 which makes it less "grabby". In fact you can control where the seam falls in my above example quite nicely by altering the distance of the z-seam hint from the object. The closer the z-seam hint point is to the object, the less likely the seam location is to be on a concave curve point. As the z-seam hint point is moved further away, the seam location is more likely to occur on a concave curve point. It's a bit of a fiddle but you can control it.

smartavionics, 2017-08-25:
I'm going to put that into a PR.

BagelOrb, 2017-08-25:
https://github.com/Ultimaker/CuraEngine/pull/589

BagelOrb, 2017-08-25:
More options on the seam: https://github.com/Ultimaker/CuraEngine/pull/590

smartavionics, 2017-09-01:
Now in the master branch, closing...

closed 2017-09-01 by smartavionics