Ultimaker/CuraEngine #581: Is there a Plugin feature? Could I make a http request with it?

Mendes11, 2017-08-09:
Hi, I have an Android App called Price My Print (feel free to download it! ;D) which i'm developing a new feature that requires the slicers to communicate with a php script when they finish their slicing work sending basic infos like fillament lengh/wheight and Extimated Time. I would like to know if there is a way of doing this with CuraEngine by some plugin feature or if it would be possible to pull request to the code something that if enabled by the user do the communication. If yes for one of the two possibilities, how would be the best way of doing it?

Thanks in advance

Appesteijn, 2017-08-09:
Hi, At the moment this is not possible to (easily) do this with CuraEngine. You could however build a Cura-plugin that does a http request when a slice is done. Look into one of the existing Cura-plugins for code examples.

Mendes11, 2017-08-09:
I'll look into it the plugins for Cura then! Thanks

Ghostkeeper, 2017-08-23:
If you were to implement this in CuraEngine, look into the `commandSocket.cpp` file and class. It has a `sendPrintTimeMaterialEstimates` function that could easily be extended to do what you need. I don't think we would accept that into our main CuraEngine code base though and CuraEngine is not pluggable.

If you were to implement this in Cura (front-end) then you could make this into a plug-in. Listen for the `printDurationMessage` signal in the `CuraEngineBackend` class.

closed 2017-08-23 by Ghostkeeper