Ultimaker/CuraEngine #585: Compiling with musl libc

lyzech, 2017-08-15:
Hi,
In order to compile CuraEngine with musl libc, I had to use this patch:
```
--- src/main.cpp.orig   2017-08-15 10:10:00.409126264 +0200
+++ src/main.cpp        2017-08-15 10:10:39.635125294 +0200
@@ -5,7 +5,9 @@
 #include <sys/time.h>
 #include <signal.h>
 #if defined(__linux__) || (defined(__APPLE__) && defined(__MACH__))
+#ifdef __GLIBC__
 #include <execinfo.h>
+#endif
 #include <sys/resource.h>
 #endif
 #include <stddef.h>
```

Is it acceptable ?
Thank you !

smartavionics, 2017-08-15:
Even with regular libc that header doesn't seem to be actually required to compile CuraEngine.

Ghostkeeper, 2017-08-29:
Acceptable. I've just removed the header since it defines only some backtrace functions. We don't even use exceptions and debug code shouldn't be in the remote repository anyway.

commit f6bdab6d (2017-08-29): Remove unused import

This is only for debugging purposes, so it shouldn't have been included in the source code.

Fixes #585.
closed 2017-08-29 by Ghostkeeper