Ultimaker/CuraEngine #588: Multi Extrusion - Prime tower only 2 colors

boelle, 2017-08-21:
Hi

can it be that the prime tower only supports 2 colors?

i use cura engine in repetier host configured for diamond hotend

i then placed the 3dbenchy and it takes some time to get all the bits placed right

i then for the testing of it picked 4 colors, blue, red, green and orange

but the prime tower only have blue and red, i looked every single layer just to be sure, but no orange or green in the prime tower

BagelOrb, 2017-08-21:
Yes. The prime tower only supports dual extrusion.

boelle, 2017-08-21:
i found another post on the same issue and i cant see why this limit has been put in there in the first place

and you dont need hardware to test it, you can just look at the gcode that comes out

BagelOrb, 2017-08-21:
Because if you print the prime tower with different colours on the same
layer, you are printing twice in the same area on the same layer. You can't
print twice in the same place. It doesn't fit.

boelle, 2017-08-21:
would it work if you just made the prime tower big enough so that the surface of each prime tower layer is big enough to fit the priming that is needed?

just a thought... or have one smaller tower for each extruder and just place them next to each other

Ghostkeeper, 2017-12-04:
See also https://github.com/Ultimaker/Cura/issues/1936.

Ghostkeeper, 2017-12-04:
Our project manager just removed this issue from our planning because it's older than 12 weeks.

boelle, 2017-12-04:
it would be like a dog chasing its own tail.... 

i will not use cura as it does not support more than 2 colors for each layer... you will not add suport for more than 2 colour pr layer since no one had a need for it...

so yes remove it from planning is the way to go

Ghostkeeper, 2017-12-04:
There's little chance that we get to spend time on it (except in our free time) unless it is applicable to Ultimaker's printers. Since Ultimaker has no printers with more than 2 extruders as yet, these sort of issues won't get fixed very quickly.

boelle, 2017-12-04:
i know

i have also given up on the multi extruder idea... 2 extruders also

while the diamond 5 way nozzle is nice there is just to much work with it and no real support form the makers of it...

2 extruders.... maybe but then again it also takes a lot of work if you are a self builder

tonydacrow, 2018-12-15:
Ladies and Gents:

Cura doesn't currently support purging of more than 2 extruders with its Prime Tower and Ooze Shield. 

Here's a way to print 3 or more colors/extruders with no need for "prime towers" or ooze shields. Simply create your own prime tower within the print!

TO EFFECTIVELY PURGE MULTI-COLOR EXTRUDERS FOR CLEAN PRINTS WITH A DIAMOND TYPE NOZZLE IN CURA: 
Create in Fusion 360 (or other appropriate program) a cylinder divided into enough pie segments to have one segment per extruder. Size the cylinder for sufficient purging (I use 30mm diameter with the same z-dimension as the tallest thing I'm printing and between 20 and 40 percent infill for 3 extruders). Make sure the cylinder prints first, i.e., before the model you actually want printed). Make sure the cylinder is big enough and the infill sufficient to extrude enough material to completely purge each extruder. Sometimes, one extruder will need more purging than the others (such as when it contains black filament or the geometry of the nozzel requires longer purges). You can always increase the infill percentage for any extruders that don't purge enough for clear colors. If that is insufficient, increase the diameter of the cylinder. That's it. A quick, simple solution that works great!

Please let me know how this works for you or if you have any questions.

Tonydacrow

closed 2019-03-19 by boelle