Ultimaker/CuraEngine #591: Recent changes allow very short extrusions to be created in infill and skin

smartavionics, 2017-08-27:
I noticed that I was getting repeatable extruder skipping at particular points in a print which didn't used to happen. The problem seems to have been caused by very short extrusions in the infill and skin that cause the extruder rate calculated by the printer firmware to come out higher than the extruder can handle. Changing LayerPlan::addExtrusionMove() to the following fixes the problem for me:
````
void LayerPlan::addExtrusionMove(Point p, const GCodePathConfig& config, SpaceFillType space_fill_type, float flow, bool spiralize, double speed_factor)
{
    if(vSize2(p - *last_planned_position) < 25)
    {
        // line is less than 5uM long, to avoid demanding an excessive extruder rate due to lack of precision
        // just travel to the line end position rather than extruding it
        addTravel_simple(p);
    }
    else
    {
        getLatestPathWithConfig(config, space_fill_type, flow, spiralize, speed_factor)->points.push_back(p);
    }
    last_planned_position = p;
}
````
The print had around a dozen lines that were either 1 or 2 um long. Other than that, all other lines were 10 um or more long. I think perhaps, having something like this would be good to guard against short lines being generated. Perhaps with debug output?

BagelOrb, 2017-08-28:
This is already done in `infill.cpp::205`.

Could you please attach a project and/or gcode file so that we can reproduce the problem?

smartavionics, 2017-08-28:
Hi Tim, can you just put in the test I show above and include some debug info when you see a short line. I'm pretty sure you will see something printed out.

BagelOrb, 2017-08-28:
I think the outline can contain short line segments.

This sounds to me like a problem which should be caught by the firmware.

Could you please describe the problem in more detail?

What's the problem exactly? Where can you see it in the gcode and what does the bad gcode command look like?

smartavionics, 2017-08-28:
What's happening is that extrusion moves are being generated that have an extremely short length (of the order of a um or 2) and so when the printer firmware calculates the extruder rate it can become very high because the rate is going to be based on the extrusion amount divided by the distance. The printer has a rate limit for the extruder but that is normally set fast enough so that the retraction/prime can be done fast. But this tiny extrude is not a retraction or prime and so the rate being asked for is above what the extrude can handle and so it skips.
Even if the printer could handle the rate, does it make sense to output lines of that length?

BagelOrb, 2017-08-28:
There seems to be a problem in `PolygonRef::simplify`, which is a very difficult function.

smartavionics, 2017-08-28:
So here's a few lines from an infill region when the above check isn't being done (I have added some distances and rates calculated from Eval/dist:
````
G1 F2250 X-1.393 Y9.878 E0.00069
M566 X60 Y60
M566 X1200 Y1200
G1 X-1.539 Y10.024 E0.00858 ; dist = 0.206 mm, rate = 3.35e-3
M566 X60 Y60
G0 F18000 X-1.408 Y9.871 ; 
M566 X1200 Y1200
G1 F2250 X-1.407 Y9.871 E0.00004  ; dist = 0.001 mm, rate = 4e-2
M566 X60 Y60
G0 F18000 X-2.623 Y8.656
M566 X1200 Y1200
G1 F2250 X-2.624 Y8.655 E0.00006 ; dist = 0.001414 mm rate = 4.24e-2
````
You can see that the lengths of the last two extrusions are tiny and the rate comes out around 10 times larger than the rate for the longer extrude.

smartavionics, 2017-08-28:
BTW, if you enable wall overlap compensation you can also get very short extrudes (although I am using my own modified version of the overlap computation so that may not be true for the stock code).

BagelOrb, 2017-08-28:
Our Marlin throws away moves which are smaller than 5 steps in all axes.

I don't think this is causing problems for us and I'm not sure whether this should be fixed in Cura or on the firmware side.

smartavionics, 2017-08-28:
Well, I honestly think it would be very poor for cura to continue to output gcode that may be impossible to execute correctly depending on the capabilities of the extruder and also the print settings used (extrusion temp, layer height, speed, etc.) Is it really up to the firmware to spot that you've asked for a completely insane extrusion rate?

BagelOrb, 2017-08-29:
Oh yeah there's the problem of extrusion rate as well of course! Sorry. I got distracted.

In our latest firmware an E-step is .0027 mm of extrusion, so the move `G1 F2250 X-1.407 Y9.871 E0.00004` is already thrown away in the firmware. I suspect the same holds for your firmware, but I don't know your exact firmware.

For our latest machine we could expect the 5 E axis steps to correspond to a XY movement in the range between .31mm - 5.8mm. (`1 / E_steps_per_mm * pi * r^2 / line_width / layer_height * 5`)

This means that the E axis will never be the limiting factor when it comes down to the 5 step rule.

---------------------------

I think your math is wrong:
```
G1 X-1.539 Y10.024 E0.00858 ; dist = 0.206 mm, rate = 0.00858 / 0.206 = 0.0417
...
G1 F2250 X-1.407 Y9.871 E0.00004  ; dist = 0.001 mm, rate = 0.00004 / 0.001 = 0.04
...
G1 F2250 X-2.624 Y8.655 E0.00006 ; dist = 0.001414 mm rate = 0.00006 / 0.001414 = 0.0424
```

BagelOrb, 2017-08-29:
I haven't really made up my mind yet. We need to discuss this some more....

smartavionics, 2017-08-29:
> I think your math is wrong:

Shit! so it its. Nonetheless, I am still getting extruder skipping when those short extrudes are included so maybe there is a problem with the firmware. I will look into that. However, even if that is true, I still think it's questionable that the slicer should output lines so short that the extrusion rate cannot be accurately specified.

BagelOrb, 2017-08-29:
I don't think the problem is in the extrusion rate.

The fact that the firmware already throws away such extrusions means that it doesn't make sense for Cura to generate them.

However, given that this doesn't seem to be a problem for our printers at the moment I think the priority of this issue is not that high for Ultimaker. However, we should accept a PR when somebody wants to improve here.

I think your initial solution proposal is a bit off.
The total amount of filament extruded shouldn't change.
You should never replace an extrusion move with a travel move.
I think we need something like this:
```
if (dist_is_too_small)
{
  if (prev_move.isTravelMove() == this_move.isTravelMove())
    remove prev_point;
  else
    remove new_point;
}
```

DDDirk, 2017-08-29:
I think the firmware should not try to outsmart the gcode. For example, I sometimes write or program gcode that uses 'insane' extrusions to filll a hole. This can be several 'normal' layers deep, which would come down to a 'layerheight' of several mm.
Not exactly what is the issue here, but the general point is that that you can never know what people may want to do (now or in the next 10 years), so it is the responsibility of who or whatever writes the gcode to make sure it makes sense.

BagelOrb, 2017-08-29:
The firmware functionality I was talking about is not trying to outsmart the gcode. It's functionality to make sure that only those commands are executed which the machine can actually execute. The 5-steps rule is in place - I think - to make sure the buffer won't overflow or drain completely.

It also has to do with the accuracy of the machine. Travel moves which are smaller than a single step of the motor can easily be thrown away. That's not the firmware outsmarting the gcode.

DDDirk, 2017-08-29:
First off, it was a general remark, based on irritiation by a printer not executing gcode (unless, for example, I add an M302 to allow 'cold' extrudes). The burden of safeguarding against mistakes should lie with the writer of the gcode, eg the slicer.

But I admit that certain decisions have to be made by the firmware, eg if a move (head or extrusion) requires less than _half_ a step. I'd say make that 1 step, because apparently someone (the programmer) or something (the slicer) wants to do something there.

BagelOrb, 2017-08-29:
> So, after quite a few hours of pouring over debug output from my printer
> firmware it looks like it could possibly have a problem that is triggered
> when there is an extrude that is so short that it would normally be
> discarded immediately preceding a non-moving retraction/prime. Just like
> this:
>
> G0 F18000 X-2.623 Y8.656

I think your firmware does not such a bad job in keeping that location in.
I think the preceding point should be removed, rather then the destination
of that short move.

I also think it would be good if Cura wouldn't generate such small moves,
but I won't put the effort in fixing this bug now because it's not a real
problem for our printers at the moment. I don't know when the Cura team
would have time for this.

I think your tweak would need a bit more love like I noted above. You
wouldn't want extrusion to start or stop at different locations.

G1 F1800 E-5

So, until the firmware gets fixed, I will just use the tweak in the first
posting in this thread.
I still think it would be a good idea if Cura did not generate such tiny
extrusion moves.

Ghostkeeper, 2017-09-03:
Maybe a line-merging technique could be implemented. Take two consecutive lines and merge them into one if they are too small.

Ghostkeeper, 2017-09-03:
I'm still of the school of thought that g-code may be executed on all printers (though with various results), and that it is therefore the responsibility of the firmware to filter out moves such as those if it won't be able to keep up the bus speed and such.

smartavionics, 2017-09-03:
Hi. In the end, the problem I was seeing turned out to be a bug in the printer firmware that was being triggered by the short extrudes. I'm still using the code above to replace all extrudes less than 5um with travel as I can see no benefit in including them. Given what you and @BagelOrb say above, I'm not bothering to take this any further.

smartavionics, 2017-09-14:
New related issue #607.