Ultimaker/CuraEngine #598: Unable to run CuraEngine.

abinashk-inf, 2017-09-01:
I compiled CuraEngine from source but was unable to run through command line. I followed instructions in
 wiki (./build/CuraEngine slice -v -j ../Cura/resources/definitions/dual_extrusion_printer.def.json -o "output/test.gcode" -e1 -s infill_line_distance=0 -e0 -l "/model_1.stl" -e1 -l "fully_filled_model.stl" ) but it failed with following error:
Loading ../Cura/resources/definitions/dual_extrusion_printer.def.json...
[ERROR] Couldn't open JSON file.
[ERROR] Failed to load json file: ../Cura/resources/definitions/dual_extrusion_printer.def.json

How can I run CuraEngine through command line?

abinashk-inf, 2017-09-02:
Hi, I was able to run the CuraEngine executable. The json file mentioned in wiki does not exist hence was causing error.

closed 2017-09-02 by abinashk-inf

Ghostkeeper, 2017-09-10:
> The json file mentioned in wiki does not exist

I'd like to fix it, but there doesn't seem to be a wiki...

ArthurKretzer, 2020-04-17:
> > The json file mentioned in wiki does not exist
> 
> I'd like to fix it, but there doesn't seem to be a wiki...

It is [here](https://github.com/Ultimaker/CuraEngine) in "Running" topic.